FreeCAD/FreeCAD-addons #5: Add FreeCADTools

lanciaux, 2016-02-04:
Hello,

I'm using FreeCADTools which is really fine for working with IPN, SHS, RHS beams and small stuff : it would be nice to have it added to the addons as well as others. The dynamics beams should be extended to others beams or BOLTS should take ideas from it.

https://github.com/Siardeni/FreeCADTools

Thank you very much

Ludovic (Belgium)

yorikvanhavre, 2016-03-07:
Unfortunately this workbench still has some problems, most texts are in french and there are many "hard-coded" paths, that will probably give a lot of problems to many people trying it, who will probably come complain here... To avoid that, I would prefer to wait a little bit until this workbench maturates a bit more. Let's keep an eye on it though, and leave this issue open so we don't forget about it.

lanciaux, 2016-03-07:
Good afternoon,

i know about portability - I myself have had some issues and I try to be in
contact with author to improve it, I have a lot of suggestion to make this
tool a metal structure builder

Yours

Ludovic (Belgium)

yorikvanhavre, 2016-03-07:
Great! A good metal construction workbench would be an important addition to FreeCAD

ManticoreII, 2016-06-07:
The hard coded paths seem to be easy enough to fix (at least for Linux). I'm currently translating it to English - might take a while to do a good job.

yorikvanhavre, 2017-01-14:
This workbench development seems abandoned, while the flamingo workbench does the same thing and I believe is now much better. So I'll close this issue for now, please reopen if needed.

closed 2017-01-14 by yorikvanhavre