FreeCAD/FreeCAD-addons #8: document python-git dependency

xzcvczx, 2016-04-11:
I installed the addons-installer macro and had it sitting there "downloading addons list" for way too long, Decided to try again and randomly clicked install/update which it then came with "python-git not found" I believe that is most likely why it is unable to download the list, but i didn't see documentation for this in the readme, And it required clicking install/update to even see that error

xzcvczx, 2016-04-12:
Actually its due to a certificate verification error so this issue really has nothing to do with what looks like optional python-git

BLCKPSTV, 2016-04-26:
I have this to, what's the solution for this?

xzcvczx, 2016-04-27:
I hacked the script to ignore ssl certificate errors. Although this is not recommended at all. It seems that python needs to be advised to use the system root certs or a bundle of root certs included in the distribution (could use the ones provided by libcurl) but even that is not really the best solution due to lack of updates

BLCKPSTV, 2016-04-27:
Well, I just dl the specific macro and installed that one only. That worked for me so I continued 🎱 .

mghansen256, 2016-08-18:
Hi @xzcvczx , could you share your insights of how to go around the certificate validation problem? That would be great.

manholo, 2017-02-05:
I commented my particular solution (on a Mac) in issue 11

yorikvanhavre, 2017-05-31:
Closing this issue as it is continued in #11

closed 2017-05-31 by yorikvanhavre