FreeCAD/FreeCAD-addons #11: No download of the addons list...

dl9sec, 2016-06-29:
The macro doesn't seem to work neither with FreeCAD 0.16, nor 0.17pre, neither on x86 nor on x64.
It starts, shows the Little GUI "Addons installer" and says "Downloading addons list..." but nothing more happens. I have a direct internet connection and the repository links within the macro are working well in my browser.

The macro source says, that there is a python-git package needed. Shouldn't it be part of FreeCAD?
How to install if not?

Any ideas?

biosafetylvl5, 2016-07-01:
The macro should be able to download without python-git. But, since I cannot replicate your problem, it might be quickest for you to follow the instructions on [this page](http://gitpython.readthedocs.io/en/stable/intro.html) to install python-git since the normal download method isn't working for you. (python-git isn't shipped with FreeCAD, and we use it as a *optional* option for downloading, but we do have a fallback normal download method. However, the normal method doesn't seem to be working for you, hence my recommendation.)

And, as a developer, I have one request for you. Could you enable your python console (by going to *View >> Panels >> Python console*) and run the macro before you install python-git? It would be quite invaluable to have the logs so we can prevent others from running into the troubles you have encountered. 

Please let us know if this helps you, and it would be great if you have the time to post the error log.

mghansen256, 2016-08-18:
This could be due to what is stated in issue number #8 : A problem with certificate validation.

On my system, I do not see any warnings in either the Python console or in the report view of FreeCAD.

OS: Windows 7
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.16.6704 (Git)
Build type: Release
Branch: releases/FreeCAD-0-16
Hash: 0c449d7e8f9b2b1fb93e3f8d1865e2f59d7ed253
Python version: 2.7.8
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

My version of the FreeCAD add-ons macro is at this revision:
https://github.com/FreeCAD/FreeCAD-addons/commit/98e86af95fcc4b33919eb2e3bcca7542e5decc20

sgrogan, 2016-08-18:
On Windows there is a packaging problem with the needed .dll's This should be fixed in the latest 0.17 development snapshots. I will be preparing updated 0.16 installers but that is not done yet.

manholo, 2017-02-05:
This is the error I get if I type in the console the url grabbing in the update worker from the add ons script:

>>> import urllib2
>>> u = urllib2.urlopen("https://github.com/FreeCAD/FreeCAD-addons")
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/Applications/FreeCAD.app/Contents/lib/python2.7/urllib2.py", line 154, in urlopen
    return opener.open(url, data, timeout)
  File "/Applications/FreeCAD.app/Contents/lib/python2.7/urllib2.py", line 431, in open
    response = self._open(req, data)
  File "/Applications/FreeCAD.app/Contents/lib/python2.7/urllib2.py", line 449, in _open
    '_open', req)
  File "/Applications/FreeCAD.app/Contents/lib/python2.7/urllib2.py", line 409, in _call_chain
    result = func(*args)
  File "/Applications/FreeCAD.app/Contents/lib/python2.7/urllib2.py", line 1240, in https_open
    context=self._context)
  File "/Applications/FreeCAD.app/Contents/lib/python2.7/urllib2.py", line 1197, in do_open
    raise URLError(err)
urllib2.URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:590)>

So I infer issue #8 should be my problem (FreeCAD .16, OSX 10.11.6, python 2.7.11)

manholo, 2017-02-05:
This seems to make it work in my mac, it possibly does the trick on other platforms too:

Modify addons_instaler.FCMacro, method run() of UpdateWorker, to use the default context. Instead of just
<<<
        import ssl
        self.progressbar_show.emit(True)
        self.info_label.emit("Downloading addons list...")
        u = urllib2.urlopen("https://github.com/FreeCAD/FreeCAD-addons")
===
        import ssl
        self.progressbar_show.emit(True)
        self.info_label.emit("reading git addons repo...")
        ctx = ssl.create_default_conext(ssl.Purpose.CLIENT_AUTH)
        u = urllib2.urlopen("https://github.com/FreeCAD/FreeCAD-addons", context=ctx)
>>>

The same should be done for all other instances of urllib2.urlopen(), or other hack to the effect.

yorikvanhavre, 2017-02-05:
This seems to work for me on linux... Any windows user who had the problem above could test if the following code snippet works without error? Just copy & paste into a python interpreter window...
```
import ssl, urllib2
ctx = ssl.create_default_context(ssl.Purpose.CLIENT_AUTH)
u = urllib2.urlopen("https://github.com/FreeCAD/FreeCAD-addons", context=ctx)
print(u.read())
```

Cody-G, 2017-02-13:
That code snippet works for me on windows.  I didn't try the workaround though.

simonvanderveldt, 2017-04-01:
Interesting. According to [PEP-476]( https://www.python.org/dev/peps/pep-0476/#technical-details) that introduced the change to enable cert validation by default the following should happen:
> http.client can then replace its usage of ssl._create_stdlib_context with the ssl._create_default_https_context .

It seems like [urllib2 actually doesn't create a default context at all](https://github.com/python/cpython/blob/master/Lib/urllib/request.py#L220) :?

@yorikvanhavre Does your fix also work with the `ssl.Purpose.SERVER_AUTH` purpose?
That's the [default](https://github.com/python/cpython/blob/master/Lib/ssl.py#L482)

yorikvanhavre, 2017-05-02:
@simonvanderveldt I have no idea :)

luzpaz, 2017-05-24:
Is https://forum.freecadweb.org/viewtopic.php?f=3&t=22616 related to this issue?

sgrogan, 2017-05-24:
I don't think this is related https://forum.freecadweb.org/viewtopic.php?f=3&t=22616
The problem on windows was a packaging error on my part with ssl libs. It is fixed in 0.17.xxxx (Tell me if I'm wrong) and I thought also on 0.16.6707
On osx there is an unrelated ssl issue that from the forum is not because of FreeCAD, If necessary I can search for the posts.
On win this should be resolved. The python-git warning will persist because we do not bundle the python-git package, but the .zip fall back should work.

yorikvanhavre, 2017-05-25:
I introduced the fix above in FreeCAD commit b604b27ad It doesn't seem to harm anyway.

luzpaz, 2017-05-25:
https://github.com/FreeCAD/FreeCAD/commit/b604b27ad3bbbacfede88e32390b076ef108331f
FTFY

luzpaz, 2017-05-26:
Can someone test yorik' fix ?

skaiser, 2017-08-30:
I tried Yorik's fix, and it worked on Mac OS Sierra.

popper99, 2017-09-11:
Newbie !  Trying to install workbenches. Have installed the Plugin Manager and in that window I press the "plugin loader" in the bottom left, but nothing happens and the window is not populated.
I tried Yorik`s fix. Did not change anything.
Please help!

elanlb, 2017-12-28:
I got it working with @manholo's suggestion. @yorikvanhavre did add the fix to AddonManager.py but not addons_installer.FCMacro so I modified every occurrence of urlopen and added the context part. Right now it's in pull request #58 , but you can [download the file](https://github.com/elanlb/FreeCAD-addons/blob/49a111976efebab2f6c903fd8f509883ea6b5e5e/addons_installer.FCMacro) and replace the old one in the macros folder. It worked for me, let me know if it works for you.

elanlb, 2018-01-17:
@yorikvanhavre I think this issue can be closed. It seems to be working now that we use the default context with SSL. PR #60 expands on this by checking for the Python version (the previous Python 2 update didn't have this SSL certificate issue).

yorikvanhavre, 2018-01-17:
Yeah I think so too. Closing this then, thanks for reminding!

closed 2018-01-17 by yorikvanhavre