FreeCAD/FreeCAD-addons #21: DesignSPHysics

luzpaz, 2017-02-13:
https://github.com/ndrs92/DesignSPHysics/issues/2

yorikvanhavre, 2017-03-17:
The author doesn't seem ready to add his work here, let's wait until he says so.

ndrs92, 2017-07-13:
Hi there!

I changed a bit of the structure of the code to match https://github.com/ndrs92/DesignSPHysics/issues/2 

The DesignSPHysics.FCMacro file loads a 'DesignSPHysics' module that is meant to be placed in the Mod directory of FreeCAD's install folder.

You can add DesignSPHysics here when you want. Notify me if more changes are needed to be made.

I'm going to update the standalone installer to copy the code following the new way and update the documentation to match the new manual install procedure.

When the work is added I'll also incorporate this way of installing.

Thanks for your work!

yorikvanhavre, 2017-07-13:
Hi Andrés,
Actually what happens in FreeCAD is that all folders found in Mod are added to the sys.path. So the DesignSPHysics folder will already be part of sys.path. So "import DesignSPHysics" will not work.
The simplest way maybe is to rename your __init__.py to something else, for ex DesignSPHysics.py. Then "import DesignSPHysics" will import that file, not the folder. But that might be confusing, maybe better to use another name.
I agree this system is not very "pythonic", there are ongoing discussions to make that better...

ndrs92, 2017-07-13:
Oh! I understand. It works for me because some time in the past I manually added a bunch of FreeCAD dirs to my path.

Anyway, tomorrow in work hours I'll rename the main Python file and test with a clean VM. Thanks for the help.

ndrs92, 2017-07-14:
Hi, I updated the project structure and the installer to match the way you described.

So now in the Mod folder there is a DesignSPHysics folder with a DesignSPHysics.py file inside. 

In the other hand the DesignSPHysics.FCMacro continues to be into the Macro folder. This way, in a Vanilla ubuntu machine works perfectly.

I hope it is ready to add it now. Thank you so much.

yorikvanhavre, 2017-07-15:
Ok everything looks good! I'm adding it. Thanks for adapting your code!

yorikvanhavre, 2017-07-15:
9ae872e7e

closed 2017-07-15 by yorikvanhavre