FreeCAD/FreeCAD-addons #29: Pre-requisite checker

luzpaz, 2017-05-10:
> We wanted to wait until we have a decent pre-requisite checker, so that a user who installs it via the addon manager and tries to run it without first having installed necessary dependencies, will at least get a helpful error message. This is in the works right now

Ref: https://forum.freecadweb.org/viewtopic.php?f=18&t=21576&p=173596#p173596

luzpaz, 2017-05-13:
Another example when installing the Geodata WB there is a dependency on the animation WB:
https://forum.freecadweb.org/viewtopic.php?f=4&t=20232#p163690

luzpaz, 2017-05-13:
Another example with installing Assembly2 
https://forum.freecadweb.org/viewtopic.php?f=22&t=22442&p=174033#p174033

luzpaz, 2017-05-14:
@yorikvanhavre any feedback?

yorikvanhavre, 2017-05-15:
Yes, we could think about something like that... It could be something simple, such as adding some "PREREQUISITE.md" file to the repo, that would contain a list of other workbenches this one depends on, that the addons manager could read, and install the needed ones if needed...

luzpaz, 2017-05-15:
@yorikvanhavre good idea. Sound simple and direct. 
@oliveroxtoby and @jaheyns what ideas have you thought about in this regard?

luzpaz, 2017-05-15:
@yorikvanhavre just to get the ball rolling: 
https://gist.github.com/luzpaz/2722740ce0319219511500c07cfad1f4

oliveroxtoby, 2017-05-18:
A simple text file describing dependencies between FreeCAD workbenches sounds fine to me.

When it comes to checking for external software installed on the system, like OpenFOAM, calculix, gmsh, etc, perhaps a function could be mandated in the addons which the manager would call to ask it to check its external dependencies. Or maybe safter just to make this the module's responsibility to do every time it is loaded. I'm not sure how easily it could be automated, especially checking version numbers, etc?

luzpaz, 2017-05-18:
Pivy usecase: https://forum.freecadweb.org/viewtopic.php?f=3&t=22453

looooo, 2017-05-18:
good idea!
once there was a discussion about adding metadata to the addons in the forum. (Currently I can't find the topic) Maybe it would be a good idea to add dependencies there. If I remeber correctly it was a file called __meta__.py. Things like [author, version, dependencies, doc-url,...] could be listed there.
Using pip as a crossplatform packagemanager would also solve this issues (partially, because more difficult dependencies can not be installed eg pivy). Something like paraview should simple output a warning if not installed, because such a big library has to be installed with system-package-manager or directly...

luzpaz, 2017-05-19:
Example of QGIS's metadata.txt file within each QGIS plugin:
https://github.com/sourcepole/qgis-openlayers-plugin/blob/master/openlayers/metadata.txt
Though it only has a min version for QGIS

joha2, 2017-05-20:
For the scipy test I wrote something simple in within the pyrate workbench. There is a paths.txt file where to check and it trys to import scipy afterwards throwing an exception if import fails. Is this the intended meaning of a pre requisite check? https://github.com/joha2/pyrate/blob/master/InitGui.py#L35 Unfortunately I had no better idea at the time being and it would be nice to have such a pre requisite checker. :-)

luzpaz, 2017-05-22:
CFDFoam guys are making some headway: https://github.com/jaheyns/CfdFoam/commit/d2f9a1f7a1be9b57154baafbb7ef2aaca4f00598

luzpaz, 2017-05-25:
We could also think about using a metadata.yml format for the file...?

luzpaz, 2017-05-25:
Linking an older thread from the forums: [The FreeCAD plugin installer](https://forum.freecadweb.org/viewtopic.php?f=10&t=15926)

luzpaz, 2017-05-25:
Played around with adapting the QGIS metadata approach: 
https://gist.github.com/luzpaz/0c83375ea503ed46d17b4b47dc35a9d9
Perhaps there should be a section appended at the bottom for dependencies like:
```
[deps]
workbenches=part,assembly,raytracer
python=3.x.x
pylibs=numpy,scipy,
3rdparty=VTK vX.y.z
3rdparty=OCC v7.1.1
FCaddon=assembly2@g3425fdr,part-o-magic@f32sed43
```

yorikvanhavre, 2017-05-25:
I would rather go by parts, because python modules will be a tricky thing to do (different installation methods on different platforms, possible permissions problems, etc). But having one WB depend on another is already something we can work on. 

Or, we could start simple, and simply pop up a dialog if some dependencies are unmet (both workbenches and python modules), and let the user sort things by him/herself. Then in a later step we see if some of those can be automated.

Your metadata.txt seems a good solution, is there an existing python parser for it?

luzpaz, 2017-05-25:
> is there an existing parser for it?
https://github.com/qgis/QGIS-Django/blob/master/qgis-app/plugins/validator.py

FYI a python implementation in the QGIS Official Python Plugin Repository code. BTW, the format is ini file so it's supported by standard libraries.
Per the folks on #QGIS

yorikvanhavre, 2017-05-25:
Ok thanks, in any case we can just start simple, with getting the lines that begin with workbenches= and pylibs=

luzpaz, 2017-05-25:
@yorikvanhavre FYI QGIS does not have dependency management yet.
Also: http://docs.qgis.org/testing/en/docs/pyqgis_developer_cookbook/plugins.html

luzpaz, 2017-05-25:
@yorikvanhavre 
Per QGIS regarding 3rd party python live
> ...you should check and tell the user what to do to install the missing deps. Or, use paver and package them as ext-libs

checkout Paver https://pythonhosted.org/Paver/

luzpaz, 2017-05-28:
The DesignSPHysics folks are taking an interesting approach to installation. Checkout:
https://github.com/ndrs92/DesignSPHysics/blob/master/installer/installer.py

luzpaz, 2017-05-29:
Started a boilerplate metadata.txt file at: https://gist.github.com/luzpaz/5e12362f0771eec9e6e82ef0e4d1ba56
```
; An example for this metadata.txt adapted from QGIS for FreeCAD
; http://docs.qgis.org/testing/en/docs/pyqgis_developer_cookbook/plugins.html#plugin-metadata
; All FreeCAD addons repos should fork this metadata.txt and adapt to their project
; the next section is mandatory

[general]
name=HelloWorld
email=me@example.com
author=Just Me
FreeCADMinimumVersion=0.17.xxxx
description=This is an example plugin for greeting the world.
    Multiline is allowed:
    lines starting with spaces belong to the same
    field, in this case to the "description" field.
    HTML formatting is not allowed.
about=This paragraph can contain a detailed description
    of the plugin. Multiline is allowed, HTML is not.
version=version 1.2
; tracker can either link to a bugtracker or a forum thread link
tracker=http://bugs.itopen.it
repository=http://www.itopen.it/repo
license=????

; end of mandatory metadata

; start of optional metadata
category=Raster
changelog=The changelog lists the plugin versions
    and their changes as in the example below:
    (also observe the two different styles of changelogs)
    2.4
    - Fixed #222: hide label if TimeManager is turned off
    - Fixed #156: copy parent style to interpolation layer
    - Fixed #109: save label settings in project
    - Fixed window resizing issues in label options gui
    - Fixed window resizing issues in video export gui
    - Fixed HiDPI issues with arch gui
    2.3.2
    - Sponsored fix for #187 and #211
    - Contributions to testing by @rduivenvoorde
    2.3.1
    - Temporarily fixed #198 sorting of frames wrong for animation?
    - Fixed #197 Export video skips frames by reverting "possible fix for #188, also implements #167? (#189)"
    2.3
    - Fixed #157: Now last animation path is saved in QSettings
    - Made archeology dialog resizable and more readable
    - Added play/pause toggle images 
    - Fixed #172: remove *.PNGw files with "Clear previous frame files in directory"
    - Improvements for WMTS support
;   or something like this 
    1.0 - First stable release
    0.9 - All features implemented
    0.8 - First testing release

; Tags are in comma separated value format, spaces are allowed within the
; tag name.
; Tags should be in English language. Please also check for existing tags and
; synonyms before creating a new one.
tags=nurbs,0.17,hello world

; these metadata can be empty, they will eventually become mandatory.
homepage=http://www.itopen.it

; icon size should be: ?????
icon=icon.png
screenshot=screenshot.png

; animated gifs are recommended
screencast=screencast.gif

; experimental flag (applies to the single version)
experimental=True

; deprecated flag (applies to the whole plugin and not only to the uploaded version)
deprecated=False

; if empty, it will be automatically set to major version + .99
FreeCADMaximumVersion=0.17.xxxx

; start of dependency section

[dependencies]
workbenches=part,assembly,raytracer

; specific python version we need to work with?
python=3.x.x

; add 3rd party oython libs
pylibs=numpy vX.y.z
pylibs=scipy vX.y.z

; add 3rd party applications
3rdparty=VTK vX.y.z
3rdparty=OCC v7.1.1

; add dependencies of other FC addons
FCaddon=assembly2@g3425fdr,part-o-magic@f32sed43

; end of dependency section
```

joha2, 2017-05-30:
Is there already code implemented which can read this metadata.txt file? What is the `category` keyword for? Best wishes
Johannes

luzpaz, 2017-05-30:
> Is there already code implemented which can read this metadata.txt file?  

@joha2 see https://github.com/FreeCAD/FreeCAD-addons/issues/29#issuecomment-304014026  
We'd need to adapt it to FreeCAD.

> What is the `category` keyword for ?  

We have yet to designate that. Most likely serves as an organization/search function to differentiate plugins/addons from each other.

yorikvanhavre, 2017-05-31:
There is now a basic dependency checker implemented in FreeCAD/FreeCAD@456d3b80f
This is how it works:
1. When pressing the "install" button, the addon manager checks if a **metadata.txt** file exists in the repo's root dir.
2. If yes, the content of that file is downloaded, and searched for lines that begin with **workbenches=** and **pylibs=**
3. All names found after workbenches= are searched in FreeCADGui.listWorkbenches().keys(). If not found there, installation will abort (a message will be printed)
4. All names found after pylibs= are tried to import. If it fails, installation will abort (a message will be printed)

luzpaz, 2017-05-31:
Sweet!
Question: What if no pylibs are necessary? Then we are aborting erroneously...?

yorikvanhavre, 2017-06-01:
Right you are. Fixed in FreeCAD/FreeCAD@58909797a

joha2, 2017-06-01:
Nice one! Is there also the possibility to divide pylibs and workbenches in mandatory and optional ones, such that the workbench or the macro could check, whether the optional ones are found and perhaps offer a failsafe solution? E.g. if you need scipy (which is problematic for win64 together with FreeCAD) and you wrote some functionality as failsafe solution for a missing scipy library then it would be nice to have a check whether these optional requirements are met.
Best wishes
Johannes

yorikvanhavre, 2017-06-01:
good idea. Done in FreeCAD/FreeCAD@99e400e87

luzpaz, 2017-06-01:
Looks like we have a new ini variables `optionalpylibs=`

HoWilgh, 2017-06-14:
To make it a little bit more complicated... does 'optional' also cover the case that at least one of the options has to be available?
For mesh generation in FEM-wb **at least one** mesher is needed.For Calculix one can use netgen or gmsh. For CFDFoam one can use gmsh or cfMesh.

HoWilgh, 2017-06-14:
BTW: CFDFoam needs gmsh>2.13, gnuplot+python-gnuplot and optional cfMesh.

luzpaz, 2017-08-04:
Another reference to addon-manager /pre-requisite chekcker: https://forum.freecadweb.org/viewtopic.php?f=10&t=23704&p=185205#p185031

sgrogan, 2017-08-04:
See starting Here: https://forum.freecadweb.org/viewtopic.php?f=18&t=23548&start=10#p183879
wmayer has given us a template!

luzpaz, 2017-11-10:
Created the beginnings of a metadata.txt generator: 
https://docs.google.com/forms/d/1-nb1ryX9zEVx1RIuuzMAvTqSr_1tebEpMIsgxAPVTYw/edit?usp=sharing

joha2, 2018-03-17:
Please have a look into #71

luzpaz, 2020-12-19:
@yorikvanhavre where is the documentation for `metadata.txt` file?

chennes, 2022-01-06:
I've added more complete dependency resolution in https://github.com/FreeCAD/FreeCAD/pull/5339 -- this reads both package.xml and metadata.txt information and consolidates them, and then walks the dependency tree when an Addon is installed an presents the user with a complete list of the package's requirements. At present it does not off to automatically install them, this is under discussion in the forums. https://forum.freecadweb.org/viewtopic.php?f=22&t=65044

You can now find rudimentary documentation of the metadata.txt file in https://wiki.freecadweb.org/Workbench_creation, and more complete documentation of package.xml in https://wiki.freecadweb.org/Package_Metadata

chennes, 2022-02-13:
That PR is now merged, I'm closing this as basically implemented (of course there's always more to do, but the basic feature is there).

closed 2022-02-13 by chennes

joha2, 2022-04-09:
@chennes Thanks for merging!