FreeCAD/FreeCAD-addons #30: Python debugger

luzpaz, 2017-05-15:
Author: @mumme74
Git repo: https://github.com/mumme74/FreeCAD/tree/editor_fixes
Forum thread: https://forum.freecadweb.org/viewtopic.php?f=10&t=21898

yorikvanhavre, 2017-05-31:
This is a copy of the FreeCAD source tree... It needs to be separated into a workbench for us tobe able to add it here.

mumme74, 2017-07-12:
Well none of these changes are setout to be a a separate entity, they are enhancements to FreeCAD core.
And coded in C++, so I don't think its related to a FreeCAD addon.

My driving force for working on this to make it easier to develop addons in the future though, when/if it makes into master.

[https://forum.freecadweb.org/viewtopic.php?f=10&t=21898&p=182194#p182194](url)

luzpaz, 2017-07-13:
@mumme74 is the debugger in it's current state considered 'stable' ?

mumme74, 2017-07-14:
@luzpaz  short answer: No, not yet.

But It sure would benefit from more testing and that wont happen without anybody using it, so....
If you have script you need debugged, please test it and report issues at: [https://github.com/mumme74/FreeCAD/issues](url)

luzpaz, 2018-06-25:
@mumme74 any progress on this exciting feature?

mumme74, 2018-08-19:
Sorry for the late response.
No havent got around to work any more on FreeCAD for a while
Sad to say that will probably not change nearterm. Life and other commitments got inte the way.

If anybody would want to build on this, they are more than welcome to take over the project.

luzpaz, 2018-08-20:
@mumme74 sorry to hear. Perhaps you can make a write up of what youve currently accomplished and what is left to do? That way, you provide some context and orientation to someone who might come along and be captivated to adopt this project? The more details the better, especially your dev environment and how you test changes to the work bench.

ferranpujolcamins, 2021-01-02:
Hi. I'm new to FreeCAD development, but this is something I'd really want to see merged. How can I help?

luzpaz, 2021-01-03:
@ferranpujolcamins fork it? :smile:

ferranpujolcamins, 2021-01-03:
Yeah! I just want to know if there's some identified issues that need to be fixed besides general testing and updating the branch

mumme74, 2021-01-04:
I have recently done some more work on this.
Please do fork !

Im glad if somebody could help/drive this thing to completion.

It is definietly not at a mergable state yet.
It needs to touch to much of FreeCAD Base, App and Gui. (not a addon, but would probably help with future development of addons)

I recently added common syntaxhighlighter (through Kate syntaxhighlighter lib)
I am currently trying to make editors switch funtionality based on plugins. Negate the need to have 2 separete texteditors (PythonEditor and TextEditor)

If I find the time I would like to make this as a standalone process. Only the python debugger running in FreeCAD main app.
That thing is currently in src/Main/IDE

luzpaz, 2021-01-05:
This can also be discussed at https://github.com/mumme74/FreeCAD/discussions/14

Eldo420, 2024-06-04:
To help solve an issue related to integrating a Python debugger, we need to get into the specifics of what the issue is. However, based on your general request concerning a Python debugger in FreeCAD, let’s consider the steps to integrate such functionality using Visual Studio Code (VSCode), a popular IDE with extensive debugging capabilities.

### Step-by-Step Guide to Set Up and Use a Python Debugger in FreeCAD via VSCode

#### 1. **Install VSCode and Python Extension**
- Download and install [VSCode](https://code.visualstudio.com/).
- Install the Python extension for VSCode. You can find it in the Extensions view by searching for "Python".

#### 2. **Configure FreeCAD for VSCode**
- Ensure FreeCAD is installed and you know the path to its executable.
- Open VSCode and create or open a workspace where your FreeCAD scripts are located.

#### 3. **Create a Launch Configuration**
- In VSCode, open the Command Palette (`Ctrl+Shift+P` or `F1`).
- Type `Debug: Open launch.json` and select it to create a launch.json file. This file will configure the debugger settings.
- Use the following template to set up the launch configuration:
```json
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "FreeCAD Debug",
            "type": "python",
            "request": "launch",
            "program": "${workspaceFolder}/your_script.py", // Path to your script
            "console": "integratedTerminal",
            "env": {
                "PYTHONPATH": "path/to/FreeCAD/lib"  // Update this with the actual path to FreeCAD's libraries
            },
            "args": [],
            "justMyCode": true
        }
    ]
}
```
Replace `"path/to/FreeCAD/lib"` with the path to your FreeCAD libraries. On Windows, it might be something like `"C:/Program Files/FreeCAD 0.19/lib"`.

#### 4. **Sample Python Script for FreeCAD**
- Create a sample Python script (e.g., `your_script.py`) for testing the debugger:
```python
import FreeCAD
import FreeCADGui
import Part
import sys

# If inside FreeCAD, close the document
if 'Unnamed' in FreeCAD.listDocuments():
    FreeCAD.closeDocument('Unnamed')

# Create a new document
doc = FreeCAD.newDocument('DebugTest')

# Add a Part Box
box = doc.addObject("Part::Box", "myBox")
doc.recompute()

print("Box created successfully.")
sys.exit()  # Exit to close FreeCAD session if running standalone
```

#### 5. **Set Breakpoints**
- Open your script in VSCode.
- Click to the left of the line number where you want to set a breakpoint. You will see a red dot indicating the breakpoint.

#### 6. **Run the Debugger**
- Go to the Debug view in VSCode (`Ctrl+Shift+D`).
- Select "FreeCAD Debug" from the dropdown menu and click the green play button to start debugging.
- VSCode should launch FreeCAD, run your script, and pause at the set breakpoints.

#### 7. **Inspect Variables and Execution**
- Once the debugger pauses at a breakpoint, you can:
  - Hover over variables to inspect their values.
  - Use the Debug toolbar to step over, step into, or step out of functions.
  - View the call stack, watch variables, and evaluate expressions in the Debug panel.

#### 8. **Debug FreeCAD Macros**
- To debug macros or scripts triggered inside FreeCAD, a different approach might be necessary. You can execute your script from within the FreeCAD Python console after setting up a remote debugger.
  - For example, using the `ptvsd` library (Python Tools for Visual Studio Debugging):
```python
import ptvsd
ptvsd.enable_attach(address=('localhost', 5678), redirect_output=True)
print("Waiting for debugger attach...")

# Wait for VSCode to attach the debugger
ptvsd.wait_for_attach()

# Your FreeCAD-script content here...
import FreeCAD
import Part

# Create a new document
doc = FreeCAD.newDocument()
box = Part.makeBox(10, 10, 10)
Part.show(box)
doc.recompute()

print("Box created successfully.")
```
- Then, in VSCode, set up a remote attach configuration in `launch.json`:
```json
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Attach to FreeCAD",
            "type": "python",
            "request": "attach",
            "connect": {
                "host": "localhost",
                "port": 5678
            },
            "pathMappings": [
                {
                    "localRoot": "${workspaceFolder}",
                    "remoteRoot": "."
                }
            ],
            "justMyCode": true
        }
    ]
}
```
- Run the macro in FreeCAD and then start the "Attach to FreeCAD" debugger in VSCode.

### Summary

By following these steps, you can set up a powerful debugging environment for FreeCAD using VSCode. This enables you to set breakpoints, inspect variables, and control execution flow effectively, making it easier to develop and troubleshoot FreeCAD scripts and add-ons.