FreeCAD/FreeCAD-addons #37: Addon Manager needs a side window to display images/animated gifs

luzpaz, 2017-05-30:
Will help users understand what the addon does when they browse through the list of addons. Screenshots and screencasts will Ben designated according to each addons metadata.txt 
`screenshot=` and `screencast=` configs

luzpaz, 2017-05-30:
https://github.com/FreeCAD/FreeCAD-addons/issues/29

luzpaz, 2017-11-12:
Created forum thread with Mockup: https://forum.freecadweb.org/viewtopic.php?f=8&t=25328

luzpaz, 2019-06-13:
@yorikvanhavre presented a new UI in https://forum.freecadweb.org/viewtopic.php?f=9&t=36556&start=10#p313848 
This was added to master in http://github.com/FreeCAD/FreeCAD/commit/3e65306a9

Testers are welcome

pgilfernandez, 2019-06-14:
May I suggest one for PieMenu?:
[image]
Original svg (you are free to modify it if you feel like):
[pie_menu_v01.svg.zip](https://github.com/FreeCAD/FreeCAD-addons/files/3290076/pie_menu_v01.svg.zip)

Cheers

luzpaz, 2019-06-14:
@pgilfernandez nice! I've posted to @triplus on the issue I opened (https://github.com/triplus/PieMenu/issues/7#issuecomment-502071747), since it really should be his call.

luzpaz, 2019-08-30:
So none of triplus's addons will get a logo, per his prerogative. 
We're still pending on:  
* https://github.com/DualSPHysics/DesignSPHysics/issues/14  
* https://github.com/edwardvmills/Silk/issues/10  
* https://github.com/microelly2/freecad-nurbs/issues/7  
* https://github.com/looooo/freecad.frametools/issues/3

luzpaz, 2019-08-30:
Follow-up to previous post...here's a quick visual progress report:  
[image]

chennes, 2022-01-01:
Note that it is no longer necessary to submit an icon for addition to the main FreeCAD repo: adding a package.xml metadata file with the path to the workbench's icon is sufficient for the Addon Manager to download and display your icon. See details at https://wiki.freecadweb.org/Package_Metadata

luzpaz, 2022-01-05:
Great. In that case I need to change this in the documentation. I just need to remember where... 😂 🤦

luzpaz, 2022-01-05:
@chennes is there an example of metadata file that you can link me to?

chennes, 2022-01-05:
There are two examples at the bottom of the wiki page about package metadata, or you can look at the Curves WB. 

https://wiki.freecadweb.org/Package_Metadata

https://github.com/tomate44/CurvesWB

chennes, 2024-03-15:
A status update: as of this writing, the following Workbenches do not have icons (in some cases the icon is present and there is simply a problem with their package.xml file so the Addon Manager can't find it)

* Better Tool Library
* bimtester
* CommandPanel
* CubeMenu
* dxf-library
* Glass
* IconThemes
* kerkythea
* Launcher
* Maker Workbench
* Navigator Indicator
* offline-documentation
* ose-piping
* persistenttoolbars
* pivy_trackers
* pluginloader
* SelectorToolbar
* ShortCuts
* SlopedPlanesMacro
* SteelColumn
* symbols_library
* TabBar
* Titleblock Workbench
* ToolbarStyle
* Trails
* workfeature-macro

Eldo420, 2024-06-04:
I can guide you step by step on how to implement the integration of workbench logos into the FreeCAD Addon Manager. This will involve several stages including acquiring logos, modifying the user interface, and coding.

### Step 1: Gather Workbench Logos

1. **Identify Workbenches**:
   - Compile a list of all the workbenches available via the Addon Manager.
   
   ```python
   workbenches = get_available_workbenches()  # Hypothetical function to list workbenches
   print(workbenches)
   ```

2. **Acquire Logos**:
   - Reach out to developers or community members to get the logos for each workbench.
   - Ensure all logos adhere to a uniform look and format, such as 64x64 pixels in PNG format.

### Step 2: Prepare the File Structure

1. **Create a Directory for Logos**:
   - Create a directory in the Addon Manager module directory to store the logo images.

   File structure:
   ```
   AddonManager/
   ├── logos/
   │   ├── workbench1.png
   │   ├── workbench2.png
   │   └── ...
   └── ...
   ```

### Step 3: Update the Addon Manager Interface

1. **Load the Logos**:
   - Write a function to load the logos from the directory.

   ```python
   import os
   from PyQt5.QtGui import QPixmap

   logos_dir = "AddonManager/logos/"

   def load_workbench_logo(workbench_name):
       logo_path = os.path.join(logos_dir, f"{workbench_name}.png")
       if os.path.exists(logo_path):
           return QPixmap(logo_path)
       return None
   ```

2. **Modify the GUI to Display Logos**:
   - Update the Addon Manager's display function to include a QLabel for the logo.

   ```python
   from PyQt5.QtWidgets import QLabel, QVBoxLayout, QWidget

   class WorkbenchDisplay(QWidget):
       def __init__(self, workbench_name, description):
           super().__init__()
           self.workbench_name = workbench_name
           self.description = description
       
           self.initUI()
       
       def initUI(self):
           self.layout = QVBoxLayout(self)
           logo_pixmap = load_workbench_logo(self.workbench_name)
           
           if logo_pixmap:
               self.logo_label = QLabel(self)
               self.logo_label.setPixmap(logo_pixmap)
               self.layout.addWidget(self.logo_label)
           
           self.description_label = QLabel(self.description, self)
           self.layout.addWidget(self.description_label)
           self.setLayout(self.layout)

   # Hypothetical integration into AddonManager's display:
   for workbench in workbenches:
       display = WorkbenchDisplay(workbench['name'], workbench['description'])
       add_display_to_main_view(display)  # Hypothetical function
   ```

### Step 4: Update the Backend

1. **Ensure Backend Supports Logo Retrieval**:
   - Modify or verify that the backend supporting the Addon Manager can retrieve the list of available logos.

   ```python
   # Ensure AddonManager backend can provide available logos if queried
   def get_available_logos():
       return [f for f in os.listdir(logos_dir) if f.endswith(".png")]

   available_logos = get_available_logos()
   print(available_logos)
   ```

### Step 5: Testing

1. **Test Across Different Systems**:
   - Ensure the Addon Manager displays logos correctly across different operating systems and screen resolutions.
   - Fix any UI/UX bugs that may arise.

### Step 6: Documentation

1. **Update Documentation**:
   - Add information about logos to the documentation for both developers and end-users.

### Additional Considerations

- **Performance**:
  - Ensure loading logos does not significantly impact the performance of the Addon Manager.
- **Fallback**:
  - Implement a fallback mechanism if a logo is not available for a specific workbench.

```python
def load_workbench_logo(workbench_name):
    logo_path = os.path.join(logos_dir, f"{workbench_name}.png")
    if os.path.exists(logo_path):
        return QPixmap(logo_path)
    else:
        return QPixmap("AddonManager/logos/default.png")  # Fallback logo
```

This approach ensures that all the workbenches are visually represented and the Addon Manager interface is improved while maintaining usability. Ensure to adapt the provided code samples according to your actual Addon Manager codebase.

closed 2025-05-05 by chennes