FreeCAD/FreeCAD-addons #38: BOLTS does not work

berndhahnebach, 2017-07-02:
Hi Yorik, 

the BOLTS package can not just copied. One needs to run some command to make a FreeCAD package out of BOLTS. Johannes has not been active for a long time. I will get in touch with Johannes, but this may take some time. Sometimes it could be cumbersome to get in touch with people who have left a project.

I have set up a BOLTS backend for Allplan and fixed some FreeCAD related stuff. I even added new data. I do not want to see BOLTS dying. On https://github.com/berndhahnebach/BOLTS/commits/boltsdev you can see my latest development in this regard. 

I made a FreeCAD package out of my latest development. On https://github.com/berndhahnebach/BOLTSFC I have set up a repository which only contains the BOLTS FreeCAD package. Means this data (the FreeCAD macro file and the directory BOLTS) have to be copied to the FreeCAD macro folder and BOLTS should run smooth. At least it does on my FreeCAD. You may could add this to the AddOn manager as long as we do not have any better.

cheers bernd

yorikvanhavre, 2017-07-04:
Hm I had forgotten that BOLTS needs to be built.
I'll have a look if it is possible to make the addons manager build it after installing, otherwise we can use your repo instead, maybe.

berndhahnebach, 2017-09-22:
How about this? Jahannes has not aswered on my mails.  :-(   Should we temporary use my repo? I will not active develop BOLTS but I would fix bugs to keep it as a usable add for FreeCAD. But this only makes sence if we use my repo with bug fixes.

bernd

yorikvanhavre, 2017-09-22:
Yes maybe it's a better idea. The original BOLTS project seems a bit stalled (last commit > 1 year), so it seems a good idea to use your "built" version in freecad... I'll do the change

closed 2017-09-22 by yorikvanhavre

berndhahnebach, 2017-09-22:
great, just tested, runs without any problems. See https://forum.freecadweb.org/viewtopic.php?f=8&t=4549&p=192787#p192787

bernd

commit cca3df4b (2017-09-22): Removed BOLTS - Added Bernd's compiled BOLTSFC instead - fixes #38