FreeCAD/FreeCAD-addons #47: Tutorial isn't informative enough

Crasher1111, 2017-09-14:
The tutorial I am following from FreeCAD isn't informative enough. For example, when it says to install the macro I have no idea how to get the macro in the first place. Its no problem, I will ask my Robotics teacher how we did it a few weeks ago but, I will have to turn in my project late. Some of us are not so tech-savy as you so please consider expanding the tutorial. Thanks.

luzpaz, 2017-09-14:
@Crasher1111 it is customary to given many details as possible when opening an issue. For example: 
* What tutorial are you using? Is it this [one](www.freecadweb.org/wiki/How_to_install_additional_workbenches)? That one does need to be cleaned up a bit.  
* What FreeCAD version are you using?  If you're using 0.16 you need to install a workbench macro manually 
it's explained in this repos' [README](https://github.com/FreeCAD/FreeCAD-addons/blob/master/README.md#installing). If you're using 0.17 then the Addon-manager is already included in to FreeCAD.
* What macro are you trying to install? Is it the Addon-Manager or is it a specific workbench that is offered through the macro?  

> I will have to turn in my project late  
Hmm...sounds like there is some sort of mis-directed laying fault going on here. Who is to blame for you turning in your project late? Sounds more like a time-management issue on your end? amirite? 
If you needed help you could of easily searched more in detail online or even better the FreeCAD forum. I know this argument well because I've used it many times :wink:
Feel free to suggest how to make the tutorial more explicit in the way you think it could on this subject. You could write a snippet submit it here and we'd consider it. 

BTW, please ask your robotics teacher to get in touch with us via the forum? We love to hear about FC used in academic contexts. Thanks!

easyw, 2017-09-14:
Maybe here

https://www.freecadweb.org/wiki/How_to_install_additional_workbenches

luzpaz, 2017-09-14:
@easyw yea, that's the one I linked to. It does need some love. I also should submit a PR to re-arrange the README for this repo. The README could have a TOC or the installation instructions could come before the descriptions of each workbench? And the descriptions could possibly be in a separate file? Something needs to change.

luzpaz, 2018-04-02:
I've updated the README (839c6d56c4285598e35684c484197da5daf9610b) so that it's more clear. I think this ticket can be closed

closed 2018-04-17 by luzpaz