FreeCAD/FreeCAD-addons #51: BOLTS is copied to the mod folder instead of the macro folder

berndhahnebach, 2017-10-10:
In the regard of https://github.com/FreeCAD/FreeCAD-addons/commit/cca3df4b382549d70531c1774f15176094d95c28

BOLTSFC is a macro, but BOLTSFC is listed at the workbenches and copied to the FreeCAD Mod folder. But it should be listed on Macros and the macro start_bolts.FCMacro and the folder BOLTSFC should be copied to the FreeCAD macro folder.

bernd

berndhahnebach, 2017-10-10:
Or should we go the other way around and change BOLTS into a Module by adding the init modules and just one tool button which starts BOLTS widget ?

bernd

elanlb, 2018-01-17:
@berndhahnebach is BOLTS being copied to the Mod folder with all of its contents or is it empty? For me, the addons installer installs macros just fine but only makes an empty folder when installing workbenches (ie BOLTS).

[image]

berndhahnebach, 2018-01-25:
No there should be a lot of data in BOLTS folder. 

ATM BOLTS is not installed propperly by the AddOn manager. BOLTS is not a workbench it is a somthing between a macro and a workbench. BOLTS has a macro which starts BOLTS. This macro should be in the FreeCAD macro folder. The BOLTS folder normaly is in the same folder like the BOLTS start macro, means the FreeCAD macro folder. This is totally different from all other macros and workbenches. ATM the AddOn installer installs it different but it seams to work anyway. 

I need to have a closer look at this BOLTS intallation by the AddOn manager. Yorik explained it quite well what we should do, install the start macro in FreeCAD macro folder and the BOLTS directory in the mod folder, and make it work properly. I hopefully will have a look at this soon. 

bernd

berndhahnebach, 2018-01-25:
It seams you are on OSX? Are you able to install other workbenches like latice2 ?

berndhahnebach, 2018-01-25:
> I think the best way would be to adapt the macro so it works with its files in Mod. Remember that all folders found in Mod are added to the python path, so after installing and restarting FreeCAD, doing something like "import BoltsSubDir" (given that BoltsSubDir is a folder inside BOLTSFC) should work.

It is exaclty what is done ATM and it works out of the box here.

closed 2018-01-25 by berndhahnebach

berndhahnebach, 2018-09-15:
> Currenty when the addons installer detects that there is one fcmacro file in the addon folder, it leaves everything in Mod, but copies the macro over to the macros folder. That allows for those "half-modules" that are not really a workbench but a bit more than macros.

The copy of the macro does not happen to be done by the addon manager, ATM see