FreeCAD/FreeCAD-addons #56: OdooPLM Plug-IN corrupted size vs previous size

OmniaGit, 2017-11-22:
Hello, 
I'm Developing a plug-in to connect odooPLM and FreeCad in order to have a proper PLM even for FreeCAD.
The plug-in works well but I gut some memory problem, FreeCAD randomly crash with the below error.
The Plug-in is made with pyqt4 not pyside could this couse this kind of problem ?
Any hint will be very appreciated.

Regards,
Matteo

*** Error in `./FreeCAD': corrupted size vs. prev_size: 0x0000555757bb3b40 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x70bcb)[0x7f6bce934bcb]
/lib/x86_64-linux-gnu/libc.so.6(+0x76f96)[0x7f6bce93af96]
/lib/x86_64-linux-gnu/libc.so.6(+0x78091)[0x7f6bce93c091]
/usr/lib/python2.7/lib-dynload/pyexpat.x86_64-linux-gnu.so(+0x89fa)[0x7f6b711c59fa]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x1397d9)[0x7f6bd71817d9]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x6861)[0x7f6bd70c93b1]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x6840)[0x7f6bd70c9390]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x86c)[0x7f6bd723215c]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x670d)[0x7f6bd70c925d]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x86c)[0x7f6bd723215c]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x670d)[0x7f6bd70c925d]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x6840)[0x7f6bd70c9390]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x86c)[0x7f6bd723215c]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x13e5b0)[0x7f6bd71865b0]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyObject_Call+0x43)[0x7f6bd711e543]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x193cbc)[0x7f6bd71dbcbc]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyObject_Call+0x43)[0x7f6bd711e543]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x194184)[0x7f6bd71dc184]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyObject_Call+0x43)[0x7f6bd711e543]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x52c0)[0x7f6bd70c7e10]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x6840)[0x7f6bd70c9390]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x6840)[0x7f6bd70c9390]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x86c)[0x7f6bd723215c]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x13e5b0)[0x7f6bd71865b0]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyObject_Call+0x43)[0x7f6bd711e543]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x193cbc)[0x7f6bd71dbcbc]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyObject_Call+0x43)[0x7f6bd711e543]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_CallObjectWithKeywords+0x47)[0x7f6bd7231587]
/usr/lib/python2.7/dist-packages/sip.x86_64-linux-gnu.so(sip_api_invoke_slot_ex+0x208)[0x7f6b71a80738]
/usr/lib/python2.7/dist-packages/PyQt4/QtCore.x86_64-linux-gnu.so(+0x1b5df2)[0x7f6b71795df2]
/usr/lib/python2.7/dist-packages/PyQt4/QtCore.x86_64-linux-gnu.so(+0x1b615d)[0x7f6b7179615d]
/usr/lib/python2.7/dist-packages/PyQt4/QtCore.x86_64-linux-gnu.so(+0x1b6d3d)[0x7f6b71796d3d]
/usr/lib/x86_64-linux-gnu/libQtCore.so.4(_ZN11QMetaObject8activateEP7QObjectPKS_iPPv+0x5fe)[0x7f6bd003e78e]
/usr/lib/x86_64-linux-gnu/libQtGui.so.4(_ZN15QAbstractButton7clickedEb+0x42)[0x7f6bd118fc72]
/usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x59a5c3)[0x7f6bd0ec65c3]
/usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x59b944)[0x7f6bd0ec7944]
/usr/lib/x86_64-linux-gnu/libQtGui.so.4(_ZN15QAbstractButton17mouseReleaseEventEP11QMouseEvent+0x74)[0x7f6bd0ec7a54]
/usr/lib/python2.7/dist-packages/PyQt4/QtGui.x86_64-linux-gnu.so(+0x54959b)[0x7f6b721d659b]
/usr/lib/x86_64-linux-gnu/libQtGui.so.4(_ZN7QWidget5eventEP6QEvent+0x290)[0x7f6bd0b46e80]
/usr/lib/python2.7/dist-packages/PyQt4/QtGui.x86_64-linux-gnu.so(+0x5a39ab)[0x7f6b722309ab]
/usr/lib/x86_64-linux-gnu/libQtGui.so.4(_ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent+0x8c)[0x7f6bd0af054c]
/usr/lib/x86_64-linux-gnu/libQtGui.so.4(_ZN12QApplication6notifyEP7QObjectP6QEvent+0x1a47)[0x7f6bd0af8ca7]
/home/mboscolo/workspace/FreeCad/lib/libFreeCADGui.so(_ZN3Gui14GUIApplication6notifyEP7QObjectP6QEvent+0x41)[0x7f6bd8175f11]
/usr/lib/x86_64-linux-gnu/libQtCore.so.4(_ZN16QCoreApplication14notifyInternalEP7QObjectP6QEvent+0x8d)[0x7f6bd0029f1d]
/usr/lib/x86_64-linux-gnu/libQtGui.so.4(_ZN19QApplicationPrivate14sendMouseEventEP7QWidgetP11QMouseEventS1_S1_PS1_R8QPointerIS0_Eb+0x14b)[0x7f6bd0af6ccb]
/usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x2461a9)[0x7f6bd0b721a9]
/usr/lib/x86_64-linux-gnu/libQtGui.so.4(_ZN12QApplication15x11ProcessEventEP7_XEvent+0xc9c)[0x7f6bd0b70b5c]
/usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x26e502)[0x7f6bd0b9a502]
/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x2a7)[0x7f6bc8d927f7]
/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x4aa60)[0x7f6bc8d92a60]
/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x2c)[0x7f6bc8d92b0c]
/usr/lib/x86_64-linux-gnu/libQtCore.so.4(_ZN20QEventDispatcherGlib13processEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE+0x64)[0x7f6bd005a854]
/usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x26e5d6)[0x7f6bd0b9a5d6]
/usr/lib/x86_64-linux-gnu/libQtCore.so.4(_ZN10QEventLoop13processEventsE6QFlagsINS_17ProcessEventsFlagEE+0x3f)[0x7f6bd00287ef]
/usr/lib/x86_64-linux-gnu/libQtCore.so.4(_ZN10QEventLoop4execE6QFlagsINS_17ProcessEventsFlagEE+0x1b5)[0x7f6bd0028b55]
/usr/lib/x86_64-linux-gnu/libQtGui.so.4(_ZN7QDialog4execEv+0xdc)[0x7f6bd0fce30c]
/usr/lib/python2.7/dist-packages/PyQt4/QtGui.x86_64-linux-gnu.so(+0x56bd45)[0x7f6b721f8d45]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x8541)[0x7f6bd70cb091]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x6840)[0x7f6bd70c9390]
[... 311 lines cut ...]
7f6bc2735000-7f6bc274f000 r-xp 00000000 08:11 666220                     /usr/lib/x86_64-linux-gnu/libsasl2.so.2.0.25
7f6bc274f000-7f6bc294e000 ---p 0001a000 08:11 666220                     /usr/lib/x86_64-linux-gnu/libsasl2.so.2.0.25
7f6bc294e000-7f6bc294f000 r--p 00019000 08:11 666220                     /usr/lib/x86_64-linux-gnu/libsasl2.so.2.0.25
7f6bc294f000-7f6bc2950000 rw-p 0001a000 08:11 666220                     /usr/lib/x86_64-linux-gnu/libsasl2.so.2.0.25
7f6bc2950000-7f6bc2964000 r-xp 00000000 08:11 3803576                    /lib/x86_64-linux-gnu/libresolv-2.24.so
7f6bc2964000-7f6bc2b63000 ---p 00014000 08:11 3803576                    /lib/x86_64-linux-gnu/libresolv-2.24.so
7f6bc2b63000-7f6bc2b64000 r--p 00013000 08:11 3803576                    /lib/x86_64-linux-gnu/libresolv-2.24.so
7f6bc2b64000-7f6bc2b65000 rw-p 00014000 08:11 3803576                    /lib/x86_64-linux-gnu/libresolv-2.24.so
7f6bc2b65000-7f6bc2b67000 rw-p 00000000 00:00 0 
7f6bc2b67000-7f6bc2b6a000 r-xp 00000000 08:11 3801293                    /lib/x86_64-linux-gnu/libkeyutils.so.1.5
7f6bc2b6a000-7f6bc2d69000 ---p 00003000 08:11 3801293                    /lib/x86_64-linux-gnu/libkeyutils.so.1.5
7f6bc2d69000-7f6bc2d6a000 r--p 00002000 08:11 3801293                    /lib/x86_64-linux-gnu/libkeyutils.so.1.5
7f6bc2d6a000-7f6bc2d6b000 rw-p 00003000 08:11 3801293                    /lib/x86_64-linux-gnu/libkeyutils.so.1.5
7f6bc2d6b000-7f6bc2d76000 r-xp 00000000 08:11 656530                     /usr/lib/x86_64-linux-gnu/libkrb5support.so.0.1
7f6bc2d76000-7f6bc2f75000 ---p 0000b000 08:11 656530                     /usr/lib/x86_64-linux-gnu/libkrb5support.so.0.1
7f6bc2f75000-7f6bc2f76000 r--p 0000a000 08:11 656530                     /usr/lib/x86_64-linux-gnu/libkrb5support.so.0.1
7f6bc2f76000-7f6bc2f77000 rw-p 0000b000 08:11 656530                     /usr/lib/x86_64-linux-gnu/libkrb5support.so.0.1
7f6bc2f77000-7f6bc2f88000 r-xp 00000000 08:11 658591                     /usr/lib/x86_64-linux-gnu/libtasn1.so.6.5.3
7f6bc2f88000-7f6bc3188000 ---p 00011000 08:11 658591                     /usr/lib/x86_64-linux-gnu/libtasn1.so.6.5.3
7f6bc3188000-7f6bc3189000 r--p 00011000 08:11 658591                     /usr/lib/x86_64-linux-gnu/libtasn1.so.6.5.3
7f6bc3189000-7f6bc318a000 rw-p 00012000 08:11 658591                     /usr/lib/x86_64-linux-gnu/libtasn1.so.6.5.3
7f6bc318a000-7f6bc31bc000 r-xp 00000000 08:11 3801570                    /lib/x86_64-linux-gnu/libidn.so.11.6.16
7f6bc31bc000-7f6bc33bc000 ---p 00032000 08:11 3801570                    /lib/x86_64-linux-gnu/libidn.so.11.6.16
7f6bc33bc000-7f6bc33bd000 r--p 00032000 08:11 3801570                    /lib/x86_64-linux-gnu/libidn.so.11.6.16
7f6bc33bd000-7f6bc33be000 rw-p 00033000 08:11 3801570                    /lib/x86_64-linux-gnu/libidn.so.11.6.16Aborted

OmniaGit, 2017-11-22:
I also have this kind of error
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x33030) [0x7f2b8266b030]
#1  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x2a568a) [0x7f2b8494568a]
#2  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x2a58da) [0x7f2b849458da]
#3  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x2a355b) [0x7f2b8494355b]
#4  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x2a3865) [0x7f2b84943865]
#5  0x7f2b83db8253 in QObject::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0xa3
#6  0x7f2b8486454c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x8c
#7  0x7f2b8486b52c in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x2cc
#8  0x7f2b8bee9f11 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/mboscolo/workspace/FreeCad/lib/libFreeCADGui.so+0x41
#9  0x7f2b83d9df1d in QCoreApplication::notifyInternal(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x8d
#10  /usr/lib/x86_64-linux-gnu/libQtCore.so.4(+0x1bdce0) [0x7f2b83dd0ce0]
#11  /usr/lib/x86_64-linux-gnu/libQtCore.so.4(+0x1baf19) [0x7f2b83dcdf19]
#12  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x2a7) [0x7f2b7cb067f7]
#13  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x4aa60) [0x7f2b7cb06a60]
#14  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x2c) [0x7f2b7cb06b0c]
#15  0x7f2b83dce854 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x64
#16  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x26e5d6) [0x7f2b8490e5d6]
#17  0x7f2b83d9c7ef in QEventLoop::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x3f
#18  0x7f2b83d9cb55 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x1b5
#19  0x7f2b83da2bd9 in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x99
#20  0x7f2b8be9fa21 in Gui::Application::runApplication() from /home/mboscolo/workspace/FreeCad/lib/libFreeCADGui.so+0x15e1
#21  ./FreeCAD(main+0x85a) [0x55e622b3a5aa]
#22  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf1) [0x7f2b826582b1]
#23  ./FreeCAD(_start+0x2a) [0x55e622b3b7ca]

luzpaz, 2017-11-23:
I'd ask in forum.freecadweb.org

luzpaz, 2018-04-16:
@OmniaGit did you get this sorted out ?

luzpaz, 2018-07-03:
@OmniaGit ping

OmniaGit, 2019-03-09:
no still persist

chennes, 2022-01-01:
I propose closing this bug report, as it's now outdated (we are now Python3/Qt5 only).

closed 2022-01-01 by luzpaz