FreeCAD/FreeCAD-addons #57: AddonManager does not work behind corporate NTLM-proxy

OctagonHex, 2017-12-18:
Der all,
There is no code to handle any proxy settings. This hampers the proliferation of FreeCAD in more traditional corporate settings.
To remedy this situation, I suggest to add:

a checkbox to turn on/off proxy support
a text field for proxy server and port
checkbox for "use authentication"
a textbox for username  (for proxy server)
a textbox for password (for proxy server)

the simplest for of adding support for the system-default proxy setting on Windows is: (this will not work for NTLM proxy, but at least "normal" proxies will work):
        proxy = urllib2.ProxyHandler()   #<--empty argument will use system-wide proxy from Windows network settings
        proxy_opener = urllib2.build_opener(proxy)
        urllib2.install_opener(proxy_opener)
        FreeCAD.Console.PrintWarning("Proxy set in UpdateWorker\n")

For NTLM-proxy support, you need to add this library: https://github.com/mullender/python-ntlm
There is an example on the page, but I have not fully succeeded in using it.

luzpaz, 2017-12-18:
@OctagonHex care to submit a PR ?

afester, 2018-01-15:
I have a PR ready at https://github.com/afester/FreeCAD-addons/tree/proxySettings which adds proxy support for the "usual" settings (no NTLM support yet):

[image]

It has a dependency on #60, I can submit the PR once #60 has been merged.

Urs-O, 2019-03-11:
I have downloaded the actual preview of Freecad and there are no proxy-settings in the download manager  - is this by design or have subsequent modifications killed that feature? Without proxy support i cant use it for corporate work.

OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16079 (Git)
Build type: Release
Branch: master
Hash: 6363c90a20b296ab69d7b52230009928199d90df
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: German/Germany (de_DE)

towe96, 2019-09-09:
Still very much relevant - no proxy settings anywhere for now, as it seems.

luzpaz, 2019-10-15:
@afester can you weigh in ?

luzpaz, 2019-10-22:
duh, we removed the file in a8a3463

Here is the file:
https://github.com/afester/FreeCAD-addons/blob/7bcb9d1e3044528300084e8eb03ff9c6b2d7d65b/addons_installer.FCMacro

What needs to happen is that we migrate this in to master.

zackles, 2020-01-08:
I coded a solution. I need to do a pull request.

zackles, 2020-01-09:
request #2877

luzpaz, 2020-01-10:
Aweseom!  
FYI, the link to the PR is https://github.com/FreeCAD/FreeCAD/pull/2877 
(you need to add the full URL to reference it since it's referencing a PR from FreeCAD/FreeCAD not FreeCAD/FreeCAD-addons

luzpaz, 2020-03-06:
@Urs-O do you mind testing the latest 0.19_pre NTLM-proxy capabilities?

Urs-O, 2020-03-13:
I will give it a try soon

Urs-O, 2020-03-13:
Tested: FreeCAD_0.19-19955-Win-Conda_vc14.x-x86_64

After switching the setting to "Use system proxy" the download of the workbench-list works. 

The download of the selected Workbench does not work. I have tested 2 of them (3D-printing_tools and ArchTextures, just randomly selected).

Traceback (most recent call last):
  File "C:\Users\obernolte\Downloads\FreeCAD_0.19-19955-Win-Conda_vc14.x-x86_64\Mod\AddonManager\addonmanager_workers.py", line 645, in run
    repo = git.Repo.clone_from(self.repos[idx][1], clonedir, branch='master')
  File "C:\Users\obernolte\Downloads\FreeCAD_0.19-19955-Win-Conda_vc14.x-x86_64\bin\lib\site-packages\git\repo\base.py", line 1017, in clone_from
    return cls._clone(git, url, to_path, GitCmdObjectDB, progress, multi_options, **kwargs)
  File "C:\Users\obernolte\Downloads\FreeCAD_0.19-19955-Win-Conda_vc14.x-x86_64\bin\lib\site-packages\git\repo\base.py", line 958, in _clone
    finalize_process(proc, stderr=stderr)
  File "C:\Users\obernolte\Downloads\FreeCAD_0.19-19955-Win-Conda_vc14.x-x86_64\bin\lib\site-packages\git\util.py", line 328, in finalize_process
    proc.wait(**kwargs)
  File "C:\Users\obernolte\Downloads\FreeCAD_0.19-19955-Win-Conda_vc14.x-x86_64\bin\lib\site-packages\git\cmd.py", line 408, in wait
    raise GitCommandError(self.args, status, errstr)
git.exc.GitCommandError: Cmd('git') failed due to: exit code(128)
  cmdline: git clone --branch=master -v https://github.com/furti/FreeCAD-ArchTextures C:\Users\obernolte\AppData\Roaming\FreeCAD\\Mod\ArchTextures
  stderr: 'Cloning into 'C:\Users\obernolte\AppData\Roaming\FreeCAD\\Mod\ArchTextures'...
fatal: unable to access 'https://github.com/furti/FreeCAD-ArchTextures/': Failed to connect to github.com port 443: Timed out
'
Traceback (most recent call last):
  File "C:\Users\obernolte\Downloads\FreeCAD_0.19-19955-Win-Conda_vc14.x-x86_64\Mod\AddonManager\addonmanager_workers.py", line 645, in run
    repo = git.Repo.clone_from(self.repos[idx][1], clonedir, branch='master')
  File "C:\Users\obernolte\Downloads\FreeCAD_0.19-19955-Win-Conda_vc14.x-x86_64\bin\lib\site-packages\git\repo\base.py", line 1017, in clone_from
    return cls._clone(git, url, to_path, GitCmdObjectDB, progress, multi_options, **kwargs)
  File "C:\Users\obernolte\Downloads\FreeCAD_0.19-19955-Win-Conda_vc14.x-x86_64\bin\lib\site-packages\git\repo\base.py", line 958, in _clone
    finalize_process(proc, stderr=stderr)
  File "C:\Users\obernolte\Downloads\FreeCAD_0.19-19955-Win-Conda_vc14.x-x86_64\bin\lib\site-packages\git\util.py", line 328, in finalize_process
    proc.wait(**kwargs)
  File "C:\Users\obernolte\Downloads\FreeCAD_0.19-19955-Win-Conda_vc14.x-x86_64\bin\lib\site-packages\git\cmd.py", line 408, in wait
    raise GitCommandError(self.args, status, errstr)
git.exc.GitCommandError: Cmd('git') failed due to: exit code(128)
  cmdline: git clone --branch=master -v https://github.com/mark1791/3D_Printing_Tools C:\Users\obernolte\AppData\Roaming\FreeCAD\\Mod\3D_Printing_Tools
  stderr: 'Cloning into 'C:\Users\obernolte\AppData\Roaming\FreeCAD\\Mod\3D_Printing_Tools'...
fatal: unable to access 'https://github.com/mark1791/3D_Printing_Tools/': Failed to connect to github.com port 443: Timed out
'

OctagonHex, 2020-03-13:
It seems the best way to solve this, could be by setting the environmental variables for the process that gets executed by the operating System: 
HTTP_PROXY
HTTPS_PROXY
or use the git config command.

https://stackoverflow.com/questions/128035/how-do-i-pull-from-a-git-repository-through-an-http-proxy

luzpaz, 2020-03-13:
CC @zackles (JIC)

zackles, 2020-03-17:
Hi guys! I will look at it in the next days. Home confinement just started today in France for at least 15 days. I won't be at work for this time. So I will not be able to test if "classical" proxy will continue to work when I do modifications to implement NTLM proxy management. I hope someone will be able to check for me.

Urs-O, 2020-03-18:
Yeah, I can check if proxy works, I will have my proxified notebook at home
😉

zackles, 2020-03-19:
@Urs-O Just to be sure :
- are you using NTLM or normal proxy?
- your problem is to have git non setted with NTLM proxy?

@OctagonHex 
You are right but I think it is better to make it work without modifying environment variable.
So I will find a way to do it with python and urllib for now (maybe later with requests).

For NTLM, I think I will go with "python-ntlm" that is a module compatible with urllib2
https://github.com/mullender/python-ntlm

luzpaz, 2020-03-20:
@zackles, please be mindful of using dependencies. We need to manage many of them in FC. If we don't have to manage yet another one, that would be ideal. You may want to discuss this on the FC forum before you run head-first in this approach.

zackles, 2020-03-20:
Created discussion here  :
https://forum.freecadweb.org/viewtopic.php?f=8&t=44363

luzpaz, 2022-01-12:
Can someone test this on the latest FreeCAD HEAD ?

chennes, 2022-01-12:
This will not work yet in 0.20, but as we migrate to using QtNetwork for communications in the AM, we will be able to add direct support for NTLM proxies without any additional libraries.

luzpaz, 2022-01-13:
AKA this will be available when we migrate to Qt6 ?

chennes, 2022-01-13:
No, I can make it work in Qt5. I hammered on it for a while tonight, but I need to discuss the ramifications on the forum. It will affect all other Qt-based network access in the code, so I want to take it slow 😃.

chennes, 2022-01-22:
Please see https://github.com/FreeCAD/FreeCAD/pull/5412 -- it should support several types of authenticated proxies, however it does not resolve the problems with Git access. Is there anyone here behind a proxy available to assist me with testing solutions?

luzpaz, 2022-01-22:
@OctagonHex 
@afester 
@Urs-O
@towe96

PR #63 (2018-02-15): Proxy settings
Partially fixes #57 (adds proxy support, but no NTLM support yet).
closed 2024-03-01 by chennes