FreeCAD/FreeCAD-addons #61: Creating a symlink on windows requires admin rights

afester, 2018-01-16:
When installing the BOLTSFC module, I am getting

```Traceback (most recent call last):
  File ".../AppData/Roaming/FreeCAD/Macro/addons_installer.FCMacro", line 665, in run
    symlink(clonedir+os.sep+f, macrodir+os.sep+f)
  File ".../AppData/Roaming/FreeCAD/Macro/addons_installer.FCMacro", line 69, in symlink
    raise ctypes.WinError()
WindowsError: [Error 1314] A required privilege is not held by the client.```

This is caused by a Windows Security Policy which restricts symbolic link creation to Administrators: https://docs.microsoft.com/en-us/previous-versions/windows/it-pro/windows-vista/cc766301(v=ws.10)#create-symbolic-links

I suggest to copy the file instead of creating a symlink on windows.

Alternatively, could modules directly be installed into the "Macro" directory, including their sub directories? Why is it required to install into "Mods" and then create a symlink of the macro file into the "Macro" directory?

luzpaz, 2018-06-25:
@sgrogan what do you think about #69 ? thanks

sgrogan, 2018-06-26:
It should be good on Win10, I don't see a path to backwards compatibility onWin.

luzpaz, 2018-07-07:
@sgrogan green light to merge the PR ?

chennes, 2022-01-22:
This has been resolved by copying the macro if the symlink fails. A long-term solution is to make the macro Manager aware of multiple macro installation locations: that's on my to-do list, but it's not a priority right now.

closed 2022-01-22 by chennes