FreeCAD/FreeCAD-addons #67: Addon-Manager should show current version of addon installed and if there is an update (show the update version

luzpaz, 2018-02-14:


luzpaz, 2018-07-15:
> Show current version of addon in _About FreeCAD_ dialog  

@DeepSOIC do any of your WBs do this already ?

DeepSOIC, 2018-07-15:
@luzpaz most likely - no. I am not quite an addon-wb expert here, though...

luzpaz, 2018-07-15:
Thanks DeepSOIC,
Perhaps @realthunder has a solution ?

realthunder, 2018-07-15:
I am not familiar with addon-manager either. But I know Stepup workbench by @easyw has the update notification function.

luzpaz, 2018-07-15:
@realthunder apologies..i should really split this issue up in to 2 tickets. I'm more interested ATM in how to hook in version information of an installed addon in to the _About FreeCAD_ dialog

chennes, 2022-01-01:
Now that we have version information in package.xml, this works as expected for displaying the current installed version. AddOns must be updated to include this metadata file, however. The same is true for showing the version of a pending update.

closed 2022-01-01 by chennes