FreeCAD/FreeCAD-addons #68: Feature: offer most up-to-date FC translations/localizations via addon manager

luzpaz, 2018-02-25:
Ref: https://forum.freecadweb.org/viewtopic.php?f=8&t=27097&p=217178#p217178
This way users don't have to wait for updates between releases.

luzpaz, 2018-04-23:
### Prerequisites
* Create github.com/FreeCAD/FreeCAD-translations
* Download Transifex CLI client called `tx` ([link](https://docs.transifex.com/client/introduction)) 
* A modified version (for FreeCAD) of this transifex translation pull [script](https://raw.githubusercontent.com/scribusproject/scribus-translations/master/update_transifex.sh)

### Proposed Workflow:
1. An admin uses the transifex pull [script](https://raw.githubusercontent.com/scribusproject/scribus-translations/master/update_transifex.sh) to download all translations and then upload them to FreeCAD/FreeCAD-translations  
2. Addon-Manager will then have access to the FreeCAD/FreeCAD-translations repo and users will be able to download said translations.

**Note**: We can also borrow from some of the musescore [script](https://github.com/musescore/tx2s3/blob/master/tx2s3.py) logic

luzpaz, 2019-07-10:
Update: FreeCAD uses Crowdin, so no need for transifex.

### Prerequisites
* Create github.com/FreeCAD/FreeCAD-translations
* Modify the [FC crowdin scripts](https://www.freecadweb.org/wiki/Crowdin_Scripts) to work with the FreeCAD bot

### Proposed Workflow:
#### Single or multiple translations
* Bot somehow polls crowdin to see if new translations have been translated
* Bot triggers script to export updated translations from crowdin and push to the GitHub repo
* User is notified via the Addon Manager that new translations are available (maybe just for dev version?)

### Side Feature
Bot can function as a way to let the devs know that translations are available and to trigger an update from Crowdin to FreeCAD master

luzpaz, 2021-05-30:
Updated workflow with info from  https://forum.freecadweb.org/viewtopic.php?f=21&t=51825&p=506882#p505457

kaktusus, 2021-05-30:
Hi.
a good plan is the basis of every action.
I have to think a little bit and get to know more. I do not know the possibilities of the crowdin system, and I do not know what tools could be used there to automate the work. being able to use cron _(crontab)_ would automate some of the work.

luzpaz, 2021-05-30:
There are a few options:
1. improve our crowdin scripts ([link](https://github.com/FreeCAD/FreeCAD/tree/master/src/Tools))
2. use the crowdin cli ([link](https://support.crowdin.com/cli-tool/))

luzpaz, 2024-09-11:
This is still a good idea, IMHO. Maybe we can make it happen post-1.0 ?