FreeCAD/FreeCAD-addons #71: Addon manager and pypi packages: any intentions to put them together?

joha2, 2018-03-17:
Heyhey,

short time ago I started a thread in the forum about refactoring an addon (pyrate) such that it is usable with PyPI. The answers were very helpful, but looo told me that this refactoring would break the usability of the package as an addon. However it is possible to install the package by PyPI, rendering a pre requisite checker obsolete, because it is done by PyPI. Is there anything planned in the future to integrate PyPI into the addon manager to make it possible to use extensions from outside of FreeCAD?

looooo, 2018-07-03:
Yes definitely. It would be easy to do so. I guess the easiest option is to write a totally new tool and integrate it into the addon-manager as another tab. I guess all we need is a file which contains all the pip installable freecad-modules.

luzpaz, 2018-07-03:
I'd suggest re-starting the above [thread](https://forum.freecadweb.org/viewtopic.php?f=22&t=27396) on the forum. Maybe even wireframing what the added tab in the Addon Manager would look like ?

looooo, 2018-07-03:
yes I will do. I think most of the things are done. https://github.com/looooo/freecad_pipintegration

looooo, 2018-07-03:
@luzpaz thanks, done https://forum.freecadweb.org/viewtopic.php?f=22&t=29584

PhoneDroid, 2025-05-26:
@looooo Is this still relevant?

PhoneDroid, 2026-04-02:
@looooo Currently not realistic, we are having trouble 
to get people to even use git to serve their addons.