FreeCAD/FreeCAD-addons #74: Window too short

joelgraff, 2018-04-07:
When opening the addon manager, the window height is too short to be useful.   I suggest setting the default height to 2/3rd's the screen height.

luzpaz, 2018-06-26:
QGIS does this:   
 
[image]

luzpaz, 2018-07-10:
@yorikvanhavre in 
https://github.com/FreeCAD/FreeCAD-addons/blob/master/addons_installer.FCMacro#L177
we see 
`self.resize(326, 304)`   
If I change that to
 `self.resize(626, 304)`  
would it make the Addon Manager widget wider ?

yorikvanhavre, 2018-07-15:
yes

luzpaz, 2018-08-02:
@joelgraff something like this? 

[image]

ref: https://forum.freecadweb.org/viewtopic.php?f=34&t=20091&p=248217#p248217

closed 2018-08-03 by yorikvanhavre