FreeCAD/FreeCAD-addons #75: No feedback after installing

joelgraff, 2018-04-07:
After installing an addon,  no feed back is given.  At a minimum,  just a message output to the report view would be useful: "addon [name] successfully installed."  Or perhaps refresh the list view to show the addon checked.

sgrogan, 2018-06-26:
For FreeCAD addons-manager, I think it's fixed.
For the addons macro, I imagine it is deprecated and won't be fixed.

luzpaz, 2018-06-26:
@sgrogan what do you mean by ?
>  For the addons macro, I imagine it is deprecated and won't be fixed.

luzpaz, 2018-08-30:
PRs welcome

luzpaz, 2018-09-07:
[image]
Looks like we do get an indication that addon installed and prompted for us to restart FreeCAD

luzpaz, 2018-09-07:
Now if the UI can give more emphasis with this information, that would be great. I'll open a separate issue for this.

closed 2018-09-07 by luzpaz