FreeCAD/FreeCAD-addons #77: UTF8 error on the macro code

anouvel, 2018-05-12:
Hi!

When I want to use the macro to list parts, I get this error:

```
Traceback (most recent call last):
  File "/Users/anouvel/Library/Preferences/FreeCAD/Macro/PartsLibrary.FCMacro", line 429, in <module>
    m.addDockWidget(QtCore.Qt.RightDockWidgetArea,ExpDockWidget(LIBRARYPATH))
  File "/Users/anouvel/Library/Preferences/FreeCAD/Macro/PartsLibrary.FCMacro", line 195, in __init__
    self.retranslateUi()
  File "/Users/anouvel/Library/Preferences/FreeCAD/Macro/PartsLibrary.FCMacro", line 200, in retranslateUi
    self.setWindowTitle(QtGui.QApplication.translate("PartsLibrary", "Parts Library", None, QtGui.QApplication.UnicodeUTF8))
<type 'exceptions.AttributeError'>: type object 'PySide2.QtWidgets.QApplication' has no attribute 'UnicodeUTF8'
```

I fixed the issue by changing each occurences of `QtGui.QApplication.UnicodeUTF8` by `0`, following this [link](https://forum.qt.io/topic/28054/unicodeutf8-not-member-qapplication-generated-with-qt5-designer).

Is it a real issue or a bad configuration from my side? My setup is a Macos with Python 2.7. Here is the configuration extracted from FreeCad.

```
OS: Mac OS X
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.13519 (Git)
Build type: Release
Branch: (HEAD detached at 1a8b868)
Hash: 1a8b868018f45ea486c0023fdbfeb06febc1fb89
Python version: 2.7.14
Qt version: 5.10.1
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: English/UnitedStates (en_US)
```

luzpaz, 2018-06-25:
Can you open a ticket on [FreeCAD-macros](https://github.com/FreeCAD/FreeCAD-macros) repo since the problem seems like it stems from the specific macro itself

luzpaz, 2018-06-26:
Opened ticket in the FreeCAD-macros repo: https://github.com/FreeCAD/FreeCAD-macros/issues/16
Closing this issue.

closed 2018-06-26 by luzpaz