FreeCAD/FreeCAD-addons #78: how to support pypi packages /new-style modules?

looooo, 2018-05-19:
It would be nice to have another tab in the addon-manager where all freecad-related packages on pip  and all new-style modules are listed and can be installed... We need this sepperated as the old way to install a addon and the new way are not compatible.
Any ideas?

luzpaz, 2018-07-09:
Related to #71

Eldo420, 2024-06-04:
The issue mentioned in FreeCAD / FreeCAD-addons #78 refers to the need to support PyPI packages or new-style modules in FreeCAD's Addon Manager. This enhancement would allow FreeCAD to easily integrate third-party libraries and modules available on the Python Package Index (PyPI), thereby expanding its functionality and usability.

Here's a step-by-step guide on how to add support for PyPI packages/new-style modules in the FreeCAD Addon Manager:

### Step 1: Add Functionality to Install PyPI Packages

1. **Update Dependencies**:
   - Ensure that `pip` is installed and accessible from within FreeCAD. This is essential for installing packages from PyPI.

   ```bash
   python -m ensurepip --upgrade
   ```

2. **Create a Function to Install PyPI Packages**:
   - Write a function that uses `pip` to install a given package.

   ```python
   import subprocess
   import sys

   def install_pypi_package(package_name):
       try:
           subprocess.check_call([sys.executable, "-m", "pip", "install", package_name])
           print(f"Package {package_name} installed successfully")
       except subprocess.CalledProcessError as e:
           print(f"Failed to install package {package_name}: {e}")
   ```

### Step 2: Modify Addon Manager to Recognize PyPI Packages

1. **Extend Configuration to Include PyPI Packages**:
   - Update the configuration or repository metadata to specify which extensions are available via PyPI.

   Example JSON snippet:
   ```json
   {
      "name": "SomeFreeCADAddon",
      "source": "github",
      "url": "https://github.com/FreeCAD/SomeFreeCADAddon",
      "description": "A useful FreeCAD Addon",
      "pypi": "somefreecadaddon"
   }
   ```

2. **Parse Configuration for PyPI Packages**:
   - Make changes to the Addon Manager to identify and handle PyPI packages.

   ```python
   import json
   import requests

   def fetch_addons_list(url):
       response = requests.get(url)
       addons_list = response.json()
       return addons_list

   def install_addon(addon):
       if 'pypi' in addon:
           install_pypi_package(addon['pypi'])
       else:
           # Existing method to handle GitHub or other sources
           install_github_addon(addon['url'])

   addons_list_url = "https://example.com/addons_list.json"
   addons_list = fetch_addons_list(addons_list_url)

   for addon in addons_list:
       print(f"Installing {addon['name']}")
       install_addon(addon)
   ```

### Step 3: Update GUI for New Install Method

1. **Modify Addon Manager Interface**:
   - Update the user interface to indicate which packages are available via PyPi and provide an appropriate installation method.

   ```python
   from PyQt5.QtWidgets import QVBoxLayout, QLabel, QPushButton, QWidget

   class AddonWidget(QWidget):
       def __init__(self, addon):
           super().__init__()
           self.addon = addon
           self.initUI()

       def initUI(self):
           layout = QVBoxLayout()
           title = QLabel(self.addon['name'], self)
           description = QLabel(self.addon.get('description', ''), self)

           install_button = QPushButton('Install', self)
           install_button.clicked.connect(self.install_addon)

           layout.addWidget(title)
           layout.addWidget(description)
           layout.addWidget(install_button)
           
           self.setLayout(layout)

       def install_addon(self):
           install_addon(self.addon)

   # Display all the addons in the Addon Manager
   for addon in addons_list:
       addon_widget = AddonWidget(addon)
       add_to_main_view(addon_widget)  # Hypothetical function to add the widget to the main view
   ```

### Step 4: Testing and Validation

1. **Test Across Different Environments**:
   - Ensure that the Addon Manager can handle installations of PyPI packages across different operating systems and Python environments.
2. **Fix Bugs**:
   - Address any issues that may arise during testing, such as missing dependencies or OS-specific errors.
3. **User Feedback**:
   - Collect feedback from users to improve the installation process and handle edge cases.

### Step 5: Documentation

1. **Update Documentation**:
   - Document the new functionality in the FreeCAD Addon Manager guide, including how to add PyPI packages to the configuration and how users can install them.
   - Provide example configuration snippets and a troubleshooting section for common issues.
   
### Additional Considerations

- **Security**:
  - Ensure that packages from PyPI are verified and trusted to avoid security vulnerabilities.
- **Dependencies**:
  - Handle any additional dependencies that PyPI packages might require.
- **Uninstallation**:
  - Implement functionality to uninstall PyPI packages if needed.

By following these steps, you can extend the FreeCAD Addon Manager to support PyPI packages, thereby making it easier to install and manage new-style modules. This enhancement will greatly benefit the FreeCAD community by providing simple access to a wide range of Python libraries.