FreeCAD/FreeCAD-addons #79: Error when installing addons

russell, 2018-05-27:
Hey,

When trying to load a part the loading bar never completes, and I see the following error in the console.

[image]

```
Traceback (most recent call last):
  File "/home/russell/.FreeCAD/Macro/addons_installer.FCMacro", line 525, in run
    desc = re.findall("<meta content=\"(.*?)\" name",p)[4]
IndexError: list index out of range
```

luzpaz, 2018-05-27:
@russell can you recreate with the latest appImage r13780 ?
https://github.com/FreeCAD/FreeCAD/releases/download/0.18_pre/FreeCAD_0.18.13780.glibc2.17-x86_64.AppImage

luzpaz, 2018-06-23:
@russell  ping

luzpaz, 2018-06-25:
No response from OP and no one else reporting a problem. Closing ticket for now

closed 2018-06-25 by luzpaz

russell, 2018-06-25:
Sorry for the late reply. Just tested it then.  Worked perfectly.

Thanks! :tada: