FreeCAD/FreeCAD-addons #80: Support pulling from Gitlab (+ other alternative git repository service providers)

luzpaz, 2018-06-23:
Per a recent discussion on the forum:. 
> Due to Microsoft buying GitHub it could happen potential future FreeCAD module developers won't join GitHub anymore. As easily as in the past. Therefore likely alternative git repository service provider support will be needed.

yorikvanhavre, 2020-11-19:
Nothing, gitlab or any others are also supported. Only detail, people without gitpython installed still rely on downloading and unpacking a zip file. This still use gihub urls iirc, it will require some changes probably

TheRaytracers, 2020-11-19:
> Nothing, gitlab or any others are also supported. Only detail, people without gitpython installed still rely on downloading and unpacking a zip file. This still use gihub urls iirc, it will require some changes probably

Ok, I looked through the code and if I see it right, the functions get_zip_url(), get_readme_url(), etc. in addonmanager_utilities.py has to be changed. Probably it is not possible to implement a solution for every git hosting service but support for GitLab would be a beginning.
The optimal solution of course would be to introduce something like a hoster object where the necessary informations are stored, but I think, we should do this later :)

Does anyone know a way to find out, where the repository is hosted?

luzpaz, 2020-11-19:
Lets remember also that this feature is fixed in the [Extension Manager](https://forum.freecadweb.org/viewtopic.php?f=10&t=50265), IIRC.

yorikvanhavre, 2020-11-19:
maybe each addon could optionally provide a zip url in their metadata.txt?

TheRaytracers, 2020-11-25:
> Lets remember also that this feature is fixed in the [Extension Manager](https://forum.freecadweb.org/viewtopic.php?f=10&t=50265), IIRC.

Hm, will it be part of the 0.19 release (in the daily version it isn't included so far)?
If not, I would vote for an simple solution like the metadata.txt.

> maybe each addon could optionally provide a zip url in their metadata.txt?

Is the reading of the file already implemented? I ask, because I think, without GitPython it would be service depend again what the exact path is.

yorikvanhavre, 2020-12-02:
> Is the reading of the file already implemented? I ask, because I think, without GitPython it would be service depend again what the exact path is.
Yes it is. It's just a matter of extending. I'll try to do that this week.

luzpaz, 2020-12-02:
@yorikvanhavre what about starting to consider [Extension Manager](https://forum.freecadweb.org/viewtopic.php?f=10&t=50265)?

TheRaytracers, 2020-12-02:
> _EDIT_ No need! Already implemented: https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/AddonManager/addonmanager_utilities.py#L251 Might not be too future-proof, though... We should find a way to get that more automatically

Unfortunately we already moved the repo to GitHub, but I think, I will survive this. Then we have the GitLab repo for developing and the GitHub repo for stable versions now :)

yorikvanhavre, 2020-12-02:
sorry it came late :/ in any case i'd like much to have more addons outside the github universe.. that would reforce all our system..
@luzpaz honestly i didn't test lately.. a priori why not, but the current one is a simple thing easy to maintain, while the other is much more complex and depends basically on one person... so that needs to be well assessed

chennes, 2022-01-01:
We can for the most part now pull from GitLab (and also Debian Salsa). I have some further refinements to this in mind, but the core functionality is implemented.

luzpaz, 2022-01-30:
@chennes how do we do this with the new changes to Addon Manager? (Or was this feature temporarily suspended?)

chennes, 2022-01-30:
We support GitHub, Gitlab, Famagit, and salsa.debian.org -- in fact, if the Addon specified the URL to the README in its package.xml file, we support any public git repo at all, regardless of its location.

luzpaz, 2022-01-31:
Probably something we should specify in https://wiki.freecad.org/Package_Metadata

chennes, 2022-01-31:
That's not really a "metadata" question (the documentation about the metadata is already pretty clear that you must specify the repo, and we highly recommend the README). I might tweak the page about the Addon Manager to spell that out, though, I need to work a lot on that page now.

luzpaz, 2022-01-31:
Happy to help with that. Let me know how I can!

chennes, 2022-02-13:
I'm going to close this as the core issue at hand is resolved. If there are other git hosts that we want to add it's pretty straightforward (especially if they are just hosted Gitlab instances). Those should get their own feature requests.

closed 2022-02-13 by chennes