FreeCAD/FreeCAD-addons #83: Add "AirPlaneDesign" add-on

Symbian9, 2018-06-25:
Add-on location:

- https://github.com/FredsFactory/FreeCAD_AirPlaneDesign

[image]

### Reference

- https://github.com/FredsFactory/FreeCAD_AirPlaneDesign/issues/1

luzpaz, 2018-06-26:
@Symbian9 please open a thread on the forum to encourage testing your workbench. Also if you could provide documentation for users and perhaps a screencast to make the process of understanding how your workbench works, easier. Thanks!

luzpaz, 2018-07-07:
@FredsFactory are you giving the green-light for submitting your WB to FreeCAD-addons ?

FredsFactory, 2018-07-07:
Yes...

FredsFactory, 2018-07-07:
Yes...

FredsFactory, 2018-07-07:
Yes
Fred

luzpaz, 2018-07-07:
AirPlaneDesign can now be installed via the FreeCAD addon Manager

commit 85549d3c (2018-07-07): Adding AirPlaneDesign workbench

Closes #83
closed 2018-07-07 by luzpaz