FreeCAD/FreeCAD-addons #87: Request to add new FreeCAD Defeaturing WB

easyw, 2018-07-02:
Could I kindly request that the Defeaturing workbench would be added?
https://github.com/easyw/Defeaturing_WB

reference at FreeCAD forum here
https://forum.freecadweb.org/viewtopic.php?f=9&t=29506
https://forum.freecadweb.org/viewtopic.php?f=8&t=29453

easyw, 2018-07-03:
@lupaz
This tool is working with FC stable and OCC 7.2 :smile:
No need to have the OCC 7.3
When they will be available I will add more features to the WB.

easyw, 2018-07-03:
@luzpaz 
I checked and the defeaturing tools are working (with some limitation on faces coming from bsplines) fine also in FC0.15 and FC0.16 😸

yorikvanhavre, 2018-07-05:
`git submodule add AddonName GitAddress` 

If you don't supply a name, it will take the same name as the one used on github (in this case Defeaturing_WB). The git addres you find under the "clone or download" green button on github repo pages

So in this case you would do this
`git pull`
to make sure your repo is clean, before doing anything. Then
`git submodule add Defeaturing https://github.com/easyw/Defeaturing_WB.git`
then
`git commit -m "Adding Defeaturing workbench"`
then 
`git push`

luzpaz, 2018-07-07:
merged in fa40b10 
@easyw sorry for delay, git has a learning curve :wink:

closed 2018-07-07 by luzpaz

easyw, 2018-07-07:
Thx :smile:

commit fa40b108 (2018-07-07): Adding Defeaturing workbench

Closes #87