FreeCAD/FreeCAD-addons #88: Geneva Mechanism Gui macro is cut off.

DieKatzchen, 2018-07-04:
Since the addon manager pulls from the wiki, and I don't have wiki access (and don't really need it), I can't do a pull request, but the full, correct macro is from [this post](https://forum.freecadweb.org/viewtopic.php?t=7756#p63483). I copied the missing part to the end of my local copy and it makes it work (after correcting indentation, of course). Can someone who has wiki access fix the code on the wiki?

luzpaz, 2018-07-04:
Hey @galou can you look in to this since it's more a macro issue than a WB issue. Thanks!

luzpaz, 2018-07-07:
@DieKatzchen I'm confused, what do you mean that the addon-manager pulls from the wiki?

DieKatzchen, 2018-07-08:
While trying to figure out where the macro is stored, I found a thread describing how the manager pulls from the wiki instead of a git for various reasons. The version I get using the manager cuts off at the same point as on the wiki, so I assume it's still true.

luzpaz, 2018-07-10:
@DieKatzchen do you mind retesting again ?

luzpaz, 2018-07-11:
Should be fixed per https://forum.freecadweb.org/viewtopic.php?f=22&t=29701#p243683
Thanks for the report.

closed 2018-07-11 by luzpaz

DieKatzchen, 2018-07-11:
Can confirm working. Thanks.