FreeCAD/FreeCAD-addons #90: Update link to InventorLoader addon

jmplonka, 2018-07-11:
I just finished adding colour support for converted SAT files.
The link is pointing to a rather old version of my addon.
Can you please update the link.

luzpaz, 2018-07-11:
What's the appropriate link ?

jmplonka, 2018-07-12:
please use:
https://github.com/jmplonka/InventorLoader/tree/ce7a237c9759e403a696d0cfb367d8666703e94b

BTW: what is the best way to update links on FreeCAD-addons

jmplonka, 2018-07-12:
update (fixed missing function):
https://github.com/jmplonka/InventorLoader/tree/b562205bc98cc90133d22a8f307079e6f873ef12

luzpaz, 2018-07-14:
@yorikvanhavre what is the command you use to update the submodules to most updated HEAD of each repo?

yorikvanhavre, 2018-07-15:
`git submodule foreach git pull`

luzpaz, 2018-07-28:
@yorikvanhavre didn't work for me for some reason. Could you try it and confirm it works ? If it does then I'll delete my fork and reclone again since I get many errors of broken submodules (probably because I deleted some incorrectly in the past).

luzpaz, 2018-07-29:
Running `git submodule foreach git pull` didn't actually do anything. I think because I didn't initialize the submodules beforehand. So I ran:  

`git submodule update --init --recursive`  
and I got this output: 
```
Submodule 'AirPlaneDesign' (https://github.com/FredsFactory/FreeCAD_AirPlaneDesign) registered for path 'AirPlaneDesign'
Submodule 'Autoload' (https://github.com/triplus/Autoload) registered for path 'Autoload'
Submodule 'BIM' (https://github.com/yorikvanhavre/BIM_Workbench.git) registered for path 'BIM'
Submodule 'BOLTSFC' (https://github.com/berndhahnebach/BOLTSFC.git) registered for path 'BOLTSFC'
Submodule 'CADExchanger' (https://github.com/yorikvanhavre/CADExchanger.git) registered for path 'CADExchanger'
Submodule 'Cfd' (https://github.com/qingfengxia/Cfd.git) registered for path 'Cfd'
Submodule 'CfdOF' (https://github.com/jaheyns/CfdOF.git) registered for path 'CfdOF'
Submodule 'CommandPanel' (https://github.com/triplus/CommandPanel) registered for path 'CommandPanel'
Submodule 'Curves' (https://github.com/tomate44/CurvesWB.git) registered for path 'Curves'
Submodule 'Defeaturing' (https://github.com/easyw/Defeaturing_WB.git) registered for path 'Defeaturing'
Submodule 'DesignSPHysics' (https://github.com/ndrs92/DesignSPHysics.git) registered for path 'DesignSPHysics'
Submodule 'ExplodedAssembly' (https://github.com/JMG1/ExplodedAssembly.git) registered for path 'ExplodedAssembly'
Submodule 'fcgear' (https://github.com/looooo/FCGear.git) registered for path 'FCGear'
Submodule 'GDT' (https://github.com/juanvanyo/FreeCAD-GDT.git) registered for path 'GDT'
Submodule 'IconThemes' (https://github.com/triplus/IconThemes.git) registered for path 'IconThemes'
Submodule 'InventorLoader' (https://github.com/jmplonka/InventorLoader.git) registered for path 'InventorLoader'
Submodule 'Launcher' (https://github.com/triplus/Launcher.git) registered for path 'Launcher'
Submodule 'Manipulator' (https://github.com/easyw/Manipulator.git) registered for path 'Manipulator'
Submodule 'NavigationIndicator' (https://github.com/triplus/NavigationIndicator) registered for path 'NavigationIndicator'
Submodule 'Part-o-magic' (https://github.com/DeepSOIC/Part-o-magic.git) registered for path 'Part-o-magic'
Submodule 'PieMenu' (https://github.com/triplus/PieMenu.git) registered for path 'PieMenu'
Submodule 'Reinforcement' (https://github.com/amrit3701/FreeCAD-Reinforcement) registered for path 'Reinforcement'
Submodule 'Render' (https://github.com/FreeCAD/FreeCAD-render.git) registered for path 'Render'
Submodule 'SelectorToolbar' (https://github.com/triplus/SelectorToolbar) registered for path 'SelectorToolbar'
Submodule 'ShortCuts' (https://github.com/triplus/ShortCuts.git) registered for path 'ShortCuts'
Submodule 'Silk' (https://github.com/edwardvmills/Silk.git) registered for path 'Silk'
Submodule 'SlopedPlanes' (https://github.com/caceres/SlopedPlanesMacro.git) registered for path 'SlopedPlanes'
Submodule 'TabBar' (https://github.com/triplus/TabBar.git) registered for path 'TabBar'
Submodule 'ToolbarStyle' (https://github.com/triplus/ToolbarStyle) registered for path 'ToolbarStyle'
Submodule 'WebTools' (https://github.com/yorikvanhavre/WebTools.git) registered for path 'WebTools'
Submodule 'animation' (https://github.com/microelly2/Animation.git) registered for path 'animation'
Submodule 'assembly2' (https://github.com/hamish2014/FreeCAD_assembly2.git) registered for path 'assembly2'
Submodule 'cadquery_module' (https://github.com/jmwright/cadquery-freecad-module.git) registered for path 'cadquery_module'
Submodule 'cura_engine' (https://github.com/cblt2l/FreeCAD-CuraEngine-Plugin.git) registered for path 'cura_engine'
Submodule 'drawing_dimensioning' (https://github.com/hamish2014/FreeCAD_drawing_dimensioning.git) registered for path 'drawing_dimensioning'
Submodule 'dxf-library' (https://github.com/yorikvanhavre/Draft-dxf-importer.git) registered for path 'dxf-library'
Submodule 'fasteners' (https://github.com/shaise/FreeCAD_FastenersWB.git) registered for path 'fasteners'
Submodule 'flamingo' (https://github.com/oddtopus/flamingo.git) registered for path 'flamingo'
Submodule 'frame' (https://github.com/looooo/freecad_frame.git) registered for path 'frame'
Submodule 'geodata' (https://github.com/microelly2/geodata.git) registered for path 'geodata'
Submodule 'kerkythea' (https://github.com/marmni/FreeCAD-Kerkythea.git) registered for path 'kerkythea'
Submodule 'kicadStepUpMod' (https://github.com/easyw/kicadStepUpMod.git) registered for path 'kicadStepUpMod'
Submodule 'lattice2' (https://github.com/DeepSOIC/Lattice2.git) registered for path 'lattice2'
Submodule 'nurbs' (https://github.com/microelly2/freecad-nurbs) registered for path 'nurbs'
Submodule 'library' (https://github.com/FreeCAD/FreeCAD-library.git) registered for path 'parts_library'
Submodule 'pcb' (https://github.com/marmni/FreeCAD-PCB.git) registered for path 'pcb'
Submodule 'persistenttoolbars' (https://github.com/triplus/PersistentToolbars.git) registered for path 'persistenttoolbars'
Submodule 'pluginloader' (https://github.com/microelly2/freecad-pluginloader.git) registered for path 'pluginloader'
Submodule 'pyrate' (https://github.com/mess42/pyrate.git) registered for path 'pyrate'
Submodule 'reconstruction' (https://github.com/microelly2/reconstruction.git) registered for path 'reconstruction'
Submodule 'retr3d' (https://github.com/Maaphoo/Retr3d.git) registered for path 'retr3d'
[... cut ...]
error: Server does not allow request for unadvertised object 9feccab298e38aed6fb146a37336d6e036845723
Fetched in submodule path 'ExplodedAssembly', but it did not contain 9feccab298e38aed6fb146a37336d6e036845723. Direct fetching of that commit failed.
```

**Note**: the last portion shows an error
> error: Server does not allow request for unadvertised object 9feccab298e38aed6fb146a37336d6e036845723
Fetched in submodule path 'ExplodedAssembly', but it did not contain 9feccab298e38aed6fb146a37336d6e036845723. Direct fetching of that commit failed.

Please advise

yorikvanhavre, 2018-08-01:
Strange, I think what you did is right.. I'll also try to reclone from scratch

yorikvanhavre, 2018-08-01:
Okay I got the same error as yours @luzpaz 
Apparenly something got screwed with ExplodedAssembly. I'll try to remove it entirely and re-add it

luzpaz, 2018-08-03:
@yorikvanhavre I think we should do these in separate commits:
1. revert b0f942c2d0d105b8c91a6c18942b30a964ca103a
2. remove `exploded_animation`
3. add `ExplodedAssembly`

luzpaz, 2018-08-24:
@yorikvanhavre did you initialize the submodules in your local repo and then run `git submodule foreach git pull` ?

luzpaz, 2018-08-28:
@yorikvanhavre cont.
When I originally cloned the repo, I didn't initialize the submobdules. So when I run `git submodule foreach git pull` I don't get any response. Are your submodules initialized ?

luzpaz, 2018-09-06:
@yorikvanhavre bump

luzpaz, 2018-10-03:
@yorikvanhavre hey mate, care to take a look over here for a moment ?

luzpaz, 2018-11-07:
@yorikvanhavre please take care of this

yorikvanhavre, 2018-11-07:
Sorry my github notifications don't work well... Only seeing this now.. Indeed this command seems to give problems and is apparently obsolete, now the preferred way to update all submodules at once seems to be `git pull --recurse-submodules`. I just did it on the repo, it gave me one error in cadquery_module that I managed to fix somehow entering the directory and doing `git checkout -f master`

luzpaz, 2018-11-24:
Yorik attempt to fix didn't work out completely, last week he got to work. so we should be back to updating the repo correctly from here on out

closed 2018-11-24 by luzpaz

jmplonka, 2019-08-07:
I've nearly completed all open issues.
Staged InventorLoader as "Release Candidate 2"
Please update link - Thanks.

luzpaz, 2019-08-07:
I've pulled all the latest from each addon repo (aka submodule)...you know it doesn't matter since the Addon Manager pulls the latest master of each submodule no matter what.