FreeCAD/FreeCAD-addons #92: Request for adding FreeCAD Lithophane Workbench

furti, 2018-08-27:
Is it possible to get the Lithophane Workbench (https://github.com/furti/FreeCAD-Lithophane) added to the addon repository?

The link to the topic in the FreeCAD Forum is https://forum.freecadweb.org/viewtopic.php?f=9&t=30496

luzpaz, 2018-08-27:
Hey @furti cool looking WB. Before we accept, please do the following:
- [x] Add some screenshots to the README.md   
- [ ] Add the word FreeCAD in the repo description: Something like: "FreeCAD workbench that convert an image to a Lithophane for 3D Printing"   
- [ ] Tag your repo as 'freecad' (you can do this in the same place where you alter the description (above)  
- [ ] What 3rd party dependencies does your WB require ? If so, please state them on the README.md
- [ ] Mention the FC forum thread that your users can post to for support in the README.md

furti, 2018-09-06:
@luzpaz is any additional input needed from me to get the Workbench added?

commit 0d6ddd9b (2018-09-06): Add Lithophane WB

Closes #92
Several members on the forum voted to inclusion.
closed 2018-09-06 by luzpaz