FreeCAD/FreeCAD-addons #96: Gitpython may not be installed but what if user prefers gitpython over downloading the .zip from Github

luzpaz, 2018-09-07:
If installing a WB via the Addon Manager and user doesn't have the GitPython package installed, the output will show the following:
```
No module named git
GitPython not found. Using standard download instead.
Downloading https://github.com/tomate44/CurvesWB/archive/master.zip
```
Lets give the user the option to choose if they want to have GitPython installed. 

Two ways to do this:  
1) Notify the user in a dialog that GitPython is not installed and that is optional since there are other methods to download. Nevertheless, if they are interested here are the instructions to download GitPython etc...
2) Perhaps use the soon-to-be-integrated pip install methodology to prompt the user if they want to install GitPython

sgrogan, 2018-09-07:
For the VS2013 builds it is difficult to make this work (the problem is PY2)
For PY3 with VS2015+, this will be possible per 2.

chennes, 2022-01-12:
Option 2 is going to be a reasonable path forward following the merge of https://github.com/FreeCAD/FreeCAD/pull/5376

chennes, 2022-09-21:
We now no longer depend on GitPython, so this is not relevant. You do need to have git installed, but we can't do that automatically, so the best we could do is warn the user they "need" it.

chennes, 2023-01-05:
From my point of view it's dead in the water: if you want to use git to download addons, you have to install git. I'm not sure what sort of meaningful user interface option I could provide.

closed 2023-01-05 by luzpaz