FreeCAD/FreeCAD-addons #99: Change repo for DualSPHysics

luzpaz, 2018-09-10:
New repo is at https://github.com/DualSPHysics/DesignSPHysics
ref: https://forum.freecadweb.org/viewtopic.php?f=37&t=20595&p=255517#p255486

luzpaz, 2018-12-04:
This still needs to be changed in https://github.com/FreeCAD/FreeCAD-addons/blob/master/.gitmodules#L112. 
CC @yorikvanhavre

yorikvanhavre, 2018-12-04:
Done with b77e988

yorikvanhavre, 2018-12-04:
I did this:

Change submodule path Git

    Update .gitmodules with the path to the new repository.
    Remove the corresponding line from the ".git/config" file.
    Delete the corresponding directory in the ".git/modules/external" directory.
    Delete the checked out submodule directory itself (unsure if this is necessary)
    Run git submodule init and git submodule update.

https://stackoverflow.com/questions/913701/how-to-change-the-remote-repository-for-a-git-submodule

closed 2018-12-04 by yorikvanhavre