FreeCAD/FreeCAD-addons #101: Feature: When Addon is installed and requires 3rd Party dependencies, AddonManager doesn't clearly notify what dependencies are needed and how to acquire them

luzpaz, 2018-09-11:
Example: Restarting FC after installing the Geodata workbench.
```
$ ./FreeCAD_0.18.14582.glibc2.17-x86_64.AppImage 
FreeCAD 0.18, Libs: 0.18R14582 (Git)
�� Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2018
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Geodat WB: Cannot import module named cv2
Geodat WB: Cannot import module named gdal gdalconst
Enumeration index -1 is out of range, ignore it
Enumeration index -1 is out of range, ignore it
Enumeration index -1 is out of range, ignore it
```

luzpaz, 2018-09-15:
@microelly2 I used Geodata WB as an example.

joha2, 2019-07-28:
This means in principle every addon author has to wrap their import directives by a `try` `except ImportError` block and show some help string if the `ImportError` exception is thrown?

chennes, 2022-01-01:
If those dependencies are documented in metadata.txt, the user will be alerted to them. A further refinement to this is obviously full dependency resolution, but we don't support that yet.

chennes, 2022-01-06:
Dependency resolution is provided by https://github.com/FreeCAD/FreeCAD/pull/5339 -- there is an ongoing discussion on the forum about whether FreeCAD should attempt to install these dependencies (and if so, how to do so).

chennes, 2022-01-12:
Dependency installation is now in https://github.com/FreeCAD/FreeCAD/pull/5376

chennes, 2022-01-22:
PR merged, we now have dependency installation.

closed 2022-01-22 by chennes