FreeCAD/FreeCAD-addons #102: BOLTS macro file is not copied to Macro folder

berndhahnebach, 2018-09-15:
* the BOLTSFC directoryis propper copied into the Mod directory by the AddOnManager
* the add on manager does not give a message about a successful installation
* the Macro "start_bolts.FCMacro" is not copied into macro directory.

bernd

for reference see https://github.com/FreeCAD/FreeCAD-addons/issues/51

luzpaz, 2018-09-19:
Not sure what I can do here. Any suggestions? 
@yorikvanhavre ithink is on vacation, FYI.

berndhahnebach, 2018-09-21:
Yes he is. https://forum.freecadweb.org/viewtopic.php?f=8&t=7448&p=254182#p254182

yorikvanhavre, 2018-10-02:
I did some more testing, and the symlinking actually works.. But there are still some problems, I had to remove the macro manually and reinstall the addon for it to work. But starting clean with boltsfc and the macro removed, it works correctly.

The main problem is that the macro is not removed when uninstalling. I'll have a look at fixing that too...

luzpaz, 2019-06-17:
@berndhahnebach is this still an issue?

berndhahnebach, 2019-06-17:
Not 100% sure, sometimes may be. Have you tried if it work for you?

vocx-fc, 2019-10-10:
> Another thing I want to add too is, when installing a macro, offer the user to add a toolbar button with it.

The problem with the BOLTSFC addon is that it is considered a workbench, not a simple macro. This behavior is a bit weird because it installs to `Mod/`, but it requires a macro to run it.

I mention this in the forum thread, [Addon BoltsFC can no longer be installed/updated](https://forum.freecadweb.org/viewtopic.php?p=340027#p340027).

> I would say this is a bad design decision by BOLTSFC. It is a "workbench", but to launch it you have to run a macro. This doesn't make sense. It is like this for historical reasons. The original BOLTS was created around 2013 when FreeCAD didn't have an Addon Manager, and there wasn't a standardized way of installing extensions.
>
> I think the proper way to solve this issue is to encapsulate the code of the start_bolts.FCMacro into a proper function in the workbench, and then provide a menu, toolbar and buttons. Then it will become a proper installable workbench, like BIM or Dodo, for example.

I think there are other workbenches that also behave like this, for example, the parts_library. If they are workbenches, they should have the proper workbench structure, and encapsulate their commands in their Mod/ directory. They shouldn't provide a separate macro; this creates confusion.

chennes, 2022-02-13:
I believe this is now fixed, feel free to reopen if it recurs.

closed 2022-02-13 by chennes