FreeCAD/FreeCAD-addons #104: README: Create a section on how to create a Workbench

luzpaz, 2018-09-29:


luzpaz, 2018-12-04:
CC @looooo

luzpaz, 2018-12-04:
It's not related ? 
Isn't it a template for a workbench ?
Addons are workbenches, modules, and macros... seems related to me.

looooo, 2018-12-05:
yes it's a template for a work-bench / module. But I don't want to mix this with the old way of creating workbenches, as this will lead to misunderstandings. Currently both ways are supported, but I guess we should think about how to make both ways more compatible as the old-style workbenches won't go away. Also most users that trying to code something for freecad have not enough python background to understand why new way of creating workbenches is better.
But yes, I will try to work on documentation if time allows.

pierrard, 2018-12-05:
Many thanks Lorenz  (Loooo) and other contributors for all you do for the community !!
I am slowly building a DIY router, 5 axis milling, and will very much appreciate the use of FreeCAD for the CAD and CAM once my hardware is ready (slow down by the cost of the parts now).
MANY THANKS!

luzpaz, 2018-12-05:
@looooo I see, thanks for explaining that. I didn't understand there was a 'better' preference. IMHO, we could create documentation for this pre-0.18 so any of the users that 'come in to the fold' of FreeCAD can make a more educated decision.

looooo, 2018-12-05:
@pierrard I am only doing what I need / like to work at. There are others like @luzpaz who try to move FreeCAD in the right direction. Best to thank the whole community, which include a lot of smart people fighting for a great project ; ).

@luzpaz yes you are right. I always underestimate the impact of docs : ).

luzpaz, 2019-09-16:
@joelgraff perhaps you could weigh in on workbench creation as well at some point ? (i know your todo list is yuuuuge but nonetheless, please consider thanks!!!)

joelgraff, 2019-09-16:
@luzpaz - Thanks for mentioning this.  Currently, I've [forked the starter kit](https://github.com/joelgraff/Workbench-Starterkit) and am, a little bit at a time, trying to build out  a general layout that is maintainable and scalable for small to mid-sized Python workbenches.  Specifically, I feel a workbench's functions are best divided across individual "Tasks" and code needs to be organized to reflect that. 

For example:

- `init_gui.py` should never have anything more than the command class imports and toolbar / context menu / menu layout code.
- A command class should only setup a task and clean up after it
- A base task class should do two things: initialize the task panel ui, and kick off the task-specific subclasses, where all of the custom code really belongs.

Of course, there are exceptions (like a command that simply executes an external program, or just opens a file), but for more complex tasks, I really think that sort of structure should be encouraged / enforced where possible.

Without a clear structure in place to mange the various workbench command / task use cases, we cannot write clear, useful developer documentation.  I realize it's no good trying to enforce absolute standards, but the goal is to provide a clean and well organized picture of how everything fits together rather than leave new devs to their own devices to hack together something that works, but may become unmaintainable as the workbench scope scales.

luzpaz, 2019-09-16:
@joelgraff   
I resonate with everything you wrote and I'm very glad this discussion is progressing. Especially:  
> Without a clear structure in place to mange the various workbench command / task use cases, we cannot write clear, useful developer documentation. I realize it's no good trying to enforce absolute [...]

I think I'll pin this issue so it's of priority.

luzpaz, 2021-03-10:
yikes, got to get around to doing this.

joelgraff, 2021-03-10:
I had a framework of my own cobbled together, but I think I lost track of
it, too.
*Joel Graff, P.E.*

*LinkedIn:* https://www.linkedin.com/in/joelcgraff

Eldo420, 2024-06-04:
Creating a detailed section in the README to explain how to create a Workbench for FreeCAD will help developers contribute effectively to the project. Here is an expanded guide that breaks down the process step by step:

### Step-by-Step Guide to Creating a Workbench in FreeCAD

1. **Introduction**

   This section explains the prerequisites, necessary tools, and a brief overview of what you will create.
   
   ```markdown
   ## Creating a Custom Workbench in FreeCAD

   FreeCAD provides a way to extend its functionalities through custom workbenches. This guide will walk you through the process of creating your own workbench.
   
   ### Prerequisites

   - Basic knowledge of Python
   - FreeCAD installed on your system
   - A text editor or IDE (e.g., Visual Studio Code, PyCharm)

   ```

2. **Setting Up Your Development Environment**

   Describe how to set up a development environment.

   ```markdown
   ### Setting Up Your Development Environment

   To get started with developing a custom Workbench, you need to ensure that your development environment is correctly set up. Follow the steps below:
   
   1. **Install FreeCAD**: Make sure FreeCAD is installed on your system.
   2. **Locate the `Mod` Directory**: Your custom Workbench will reside in the `Mod` directory of your FreeCAD installation. On Windows, it is usually located at `C:\Program Files\FreeCAD\Mod`. On Linux, it's typically found in `/usr/lib/freecad/Mod` or `/usr/local/lib/freecad/Mod`.
   3. **Create a New Folder for Your Workbench**: For example, create a folder named `MyWorkbench`.
   
   ```
   
3. **Basic Folder Structure**

   Outline the basic folder structure necessary for a FreeCAD Workbench.
   
   ```markdown
   ### Basic Folder Structure
   
   Your Workbench will follow a specific folder and file structure. Below is an example:
   
   ```
   MyWorkbench/
   ├── Init.py
   ├── InitGui.py
   ├── MyWorkbench.py
   ├── Resources/
       ├── icons/
       ├── translations/
   └── __init__.py
   ```
   
   - **`Init.py`**: Contains initialization code that runs when FreeCAD starts.
   - **`InitGui.py`**: Handles graphical user interface initialization.
   - **`MyWorkbench.py`**: The main file containing your workbench logic.
   - **`Resources`**: Contains icons, translation files, and other resources.
   - **`__init__.py`**: Indicates that the directory is a Python package.
   ```

4. **Creating Required Files**

   Detail the contents and purpose of each required file.

   ```markdown
   ### Creating Required Files

[... 264 lines cut ...]
Place your icons, translations, and other resources under the `Resources` directory. Use them in your tool and workbench definitions as shown in the scripts above.

### Packaging and Installing Your Workbench

- **Packaging**: Zip your `MyWorkbench` folder and its contents.
- **Installing**: Unzip the folder into FreeCAD's `Mod` directory.

When users start FreeCAD, your Workbench should now appear in the Workbench list.

### Advanced Features

- **Custom Dialogs**: Use `PySide` or `PyQt` to create custom dialogs.
- **Complex Algorithms**: Integrate third-party Python libraries to add advanced capabilities.

### Troubleshooting and Debugging

- **Log Files**: Check FreeCAD log files for any error messages.
- **Debugging**: Use Python's `pdb` module or IDE debugging tools.

### Conclusion

You've now created a basic custom Workbench for FreeCAD. Feel free to extend it with more features and share it with the community!
```

This guide should help developers create their own FreeCAD Workbench from scratch, providing all the necessary steps and code snippets.

lukh, 2024-10-13:
Hello there,
I wrote my first workbench for FreeCAD (pure python) and found it quite hard to get access to "good doc" (ie, up to date, following standard python rules, etc)
The default template give informations, but not how to create a task for instance, 
The wiki is... (sorry :s )  complicated to get to, too many pages, in different sections, (maybe the scripting page are okay, but I couldn't find infos like; where to put the code snippet, I had to figure it out with some existing WB, (but not specifically up-to-date... or Draft Workbench, that uses a lot of shared / polymorphism code, so hard to follow when you want understand how a WB works...

I want to help, I can help @joelgraff  on the template, or on the wiki; I would probably need validation that the code I wrote is good enough for that

(BTW: learning FreeCAD for one year or so, love it, works like a charm :) ),

Bests

luzpaz, 2024-10-18:
@lukh It would be very welcome update and change. Can you connect to the forum or discord chat and start a discussion relate to this?

PhoneDroid, 2026-04-02:
Will be addressed via the Addon Academy.

closed 2026-04-02 by PhoneDroid