FreeCAD/FreeCAD-addons #105: Document how Workbench creators can inject WB version in to About FreeCAD dialog

luzpaz, 2018-10-03:
The `Help > About FreeCAD` dialog should show what workbenches are installed and their respective versions. 
1. Make this possible to do via code. 
2. Document the hook that devs need to call in their own workbenches.

kbwbe, 2019-01-04:
Highly appreciated ! If there is an interface i will use it instantly.

easyw, 2019-01-05:
As a workaround, I placed a print of the WB release version which is triggered when a user switch from an other WB to one of mine... in this way the WB release is displayed at least in the Report view.
Moreover I have also a help button in each of mine WB in which there is some useful tips and the release number of the tool.

luzpaz, 2019-01-05:
@easyw  can you share the location in your code ?

easyw, 2019-01-05:
@luzpaz 
here it is the Report Panel printing:
https://github.com/easyw/Manipulator/blob/master/InitGui.py#L85

kbwbe, 2019-01-06:
Perhaps this is a way to provide the Version Numbers.

On Start of FC the constructor of each external python WB is executed, regardless it is activated or not.
If we do this:

`class a2pWorkbench (Workbench):

    def __init__(self):
        import a2plib
        self.__class__.Icon = a2plib.pathOfModule() + "/icons/a2p_workbench.svg"
        self.__class__.MenuText = 'A2plus V0.2.3' #<========================Version here....
        self.__class__.ToolTip  = 'An other assembly workbench for FreeCAD'

`
we get the version of a python WB within the WB selector visible.

luzpaz, 2019-01-13:
@kbwbe Found it in your workbench https://github.com/kbwbe/A2plus/blob/master/InitGui.py#L38-L42

> Edit: As WB selector is already showing a list of all installed WB's, this list already exists within FC Core. Someone of the C++ Guy's would to have to read this list out and put it to the FC About Dialog. (if possible)

@wwmayer would this be something you could do before the 0.18 release and I'll let as many of the addon authors know about it once you do? (similar to what I've done in #93)

luzpaz, 2019-01-16:
Looks like the temporary workaround doesn't put the version in the About dialog but instead next to the Workbench name in the dropdown  
[image]

luzpaz, 2019-01-16:
Looks like Manipulator workaround adds the version to the command line:  
[image]

Rentlau, 2019-02-23:
On my side I print the version in the console (in Activated function) but also in preferences panel.
[image]

hyarion, 2021-01-03:
I've made a proof-of-concept of this feature (based on the License tab) The PoC can be compared to master here: https://github.com/hyarion/FreeCAD/pull/new/feature-workbenches-in-about

I propose that we add new optional attributes for the missing information.
In my PoC I've added support for Version and Url, but we could add/remove to what you think would be best.

Here's a diff that adds a version and an url to the Spreadsheet workbench entry: 
```diff
diff --git a/src/Mod/Spreadsheet/InitGui.py b/src/Mod/Spreadsheet/InitGui.py
index dda6423eb..fda7882df 100644
--- a/src/Mod/Spreadsheet/InitGui.py
+++ b/src/Mod/Spreadsheet/InitGui.py
@@ -34,6 +34,8 @@ class SpreadsheetWorkbench ( Workbench ):
         self.__class__.Icon = FreeCAD.getResourceDir() + "Mod/Spreadsheet/Resources/icons/SpreadsheetWorkbench.svg"
         self.__class__.MenuText = "Spreadsheet"
         self.__class__.ToolTip = "Spreadsheet workbench"
+        self.__class__.Version = "0.12"
+        self.__class__.Url = "http://www.freecadweb.org"
 
     def Initialize(self):
         # load the module
```

If the information is missing, it will just ignore adding that data.

Here is how it looks in action:
[image]

@luzpaz and others, is this in the line of what you imagined or would you like me to change anything?
(I omitted adding author as I was thinking that this could be seen if you follow the url)

luzpaz, 2021-01-04:
This is great. But maybe taking it some steps too far:
1. Not sure we need a new tab
2. No need to version the core workbenches, just external workbenches
3. In my mind, I imagined all the necessary version info could still be show in the `About` tab
4. That we format the About info to display more info (utilize screen 'real-estate' more efficiently)
Here's a back-of-the-napkin idea:
```
OS: Manjaro Linux (KDE//usr/share/xsessions/plasma)
Word size of OS/FreeCAD (64-bit/64-bit)
Version: 0.19.23578 (Git) AppImage  Build type: Release
Branch: master                      Hash: 50c3cbf005
Python v3.8.6   Qt v5.12.5  Coin v4.0.0 OCC v7.4.0
Locale: English/United States (en_US)
External Workbenches:
A2plus v2.1
Assembly4 v2.4
Trails vAlpha
```

hyarion, 2021-01-04:
1. Ah, I now understand what you mean with inject. Technically the Library tab is injected to the UI which threw me off... Removing tab! 
2. I'll try to see if I can find a way to check if a workbench is internal or external.
3 & 4. I'm not sure what you mean, could you draw a mockup of how you would like it to look?

luzpaz, 2021-01-10:
Sorry, that was a little rushed of me. So there are 2 UI requests being asked here. One is the Qt About dialog info and the other is the `Copy to clipboard` output. For now I'm proposing to alter the latter since this is the most pressing because users can save forum member a lot of extra questions by posting exactly why version of external workbenches are instaled on their system. 

This is the current output of the `Copy to clipboard` function in the FreeCAD About dialog
```
OS: Manjaro Linux (KDE//usr/share/xsessions/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.23601 (Git)
Build type: Release
Branch: master
Hash: 294a2ebc9d56dd91c0d05155432283a1aa6d17b8
Python version: 3.9.1
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)
```
Proposal for optimizing space + inclusion of external workbenches:
```
OS: Manjaro Linux (KDE//usr/share/xsessions/plasma)
Word size of OS/FreeCAD (64-bit/64-bit)
Version: 0.19.23578 (Git) AppImage  Build type: Release
Branch: master                      Hash: 50c3cbf005
Python v3.8.6   Qt v5.12.5  
Coin v4.0.0     OCC v7.4.0
Locale: English/United States (en_US)
External Workbenches:
A2plus v2.1      Assembly4 v2.4
BIM vX.Y         SheetMetal vX.Y
Fasteners vX.Y   Trails vAlpha
```

hyarion, 2021-01-10:
@luzpaz check! That makes much more sense.
I'll modify my implementation accordingly.

chennes, 2022-02-13:
With the new package.xml metadata file it is possible for any code to inspect the installed versions of Addons (if they provide that file, of course). So there is no need to "inject", per se, the About dialog can just look it up. Over time as more Addons include this info the listing will be more complete.

luzpaz, 2022-02-13:
@chennes how is it represented in the About dialog?

chennes, 2022-02-13:
It's not -- but now it _could be_, with no further changes to anything except that dialog itself.

luzpaz, 2022-02-13:
Thanks @chennes. Therefore, next step is #199  
Closing this ticket.

closed 2022-02-13 by luzpaz