FreeCAD/FreeCAD-addons #109: freecad-daily fails to install part library

maidenone, 2018-11-20:
freecad-daily do not install python-git and fallback fails?
See: https://forum.freecadweb.org/viewtopic.php?f=3&t=32330

luzpaz, 2019-03-09:
Related #96

luzpaz, 2019-03-09:
Yorik writes:
> The parts library is huge, even with git it takes a long time to download. I suspect the http method (non-git) fails because of that, I'm not too sure what to do there...

Maidenone responds:
> I am not sure it actually failed, my issue was that the GUI was unresponsive and i got no info/feedback, the download might have been running in the background and the GUI might have become operational again when the download was done or had failed?
> 
> Maybe start the download in a different thread and hand back the GUI to the user or atleast state "download in progress"
> 
> Also, it should not be marked "installed" until after the install/download have finished, i guess the state was set when the install button was pressed. (I killed FreeCAD afer a few minutes of unresponsiveness)

luzpaz, 2019-10-23:
#120

closed 2022-01-25 by luzpaz