FreeCAD/FreeCAD-addons #112: error on timber addon

berndhahnebach, 2019-01-05:
on update of tmber add on I get the following error ...

`
GitPython not found. Using standard download instead.
Downloading https://github.com/wood-galaxy/FreeCAD-Timber/archive/master.zip
Traceback (most recent call last):
  File "/home/hugo/Documents/dev/freecad/freecadbhb_dev/build/Mod/AddonManager/AddonManager.py", line 740, in run
    u = urlopen(url)
  File "/home/hugo/Documents/dev/freecad/freecadbhb_dev/build/Mod/AddonManager/addonmanager_utilities.py", line 36, in urlopen
    u = urllib2.urlopen(url, context=ssl_ctx)
  File "/usr/lib/python3.7/urllib/request.py", line 222, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.7/urllib/request.py", line 531, in open
    response = meth(req, response)
  File "/usr/lib/python3.7/urllib/request.py", line 641, in http_response
    'http', request, response, code, msg, hdrs)
  File "/usr/lib/python3.7/urllib/request.py", line 569, in error
    return self._call_chain(*args)
  File "/usr/lib/python3.7/urllib/request.py", line 503, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.7/urllib/request.py", line 649, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 404: Not Found
`

on 

OS: Debian GNU/Linux buster/sid
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15576 (Git)
Build type: Unknown
Branch: femtmp
Hash: 0136445404fa1d45fff3d1254f938c147fc3834e
Python version: 3.7.2rc1
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: German/Switzerland (de_CH)

may be someone could confirm this

luzpaz, 2019-01-19:
Can't reproduce on 
```
OS: Manjaro Linux
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15640 (Git) AppImage
Build type: Release
Branch: master
Hash: d94fc4988e2d26ea9ef65b09cc84f1f287411181
Python version: 3.6.7
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

luzpaz, 2019-01-26:
@berndhahnebach i bet you won't be able to reproduce since the error above I think is due to github not being available at that moment ?

luzpaz, 2019-06-17:
Closing issue.. re-open if necessary

closed 2019-06-17 by luzpaz