FreeCAD/FreeCAD-addons #115: GD&T

aversario, 2019-01-18:
GDT addon will crash on linux. Tried v 0.16 and 0.17:
`./FreeCAD-0.16-6712.glibc2.17-x86_64.AppImage 
/bin/bash: /tmp/.mount_FreeCApeQQti/lib/x86_64-linux-gnu/libtinfo.so.5: no version information available (required by /bin/bash)
desktop-file-validate is missing. Skipping /tmp/.mount_FreeCApeQQti/usr/bin//freecad-0.16.wrapper.
FreeCAD 0.16, Libs: 0.16R6712 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2015
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Geometric Dimensioning & Tolerancing workbench activated
Geometric Dimensioning & Tolerancing workbench desactivated
Geometric Dimensioning & Tolerancing workbench activated
Geometric Dimensioning & Tolerancing workbench desactivated
Geometric Dimensioning & Tolerancing workbench activated
freecad-0.16: SoAsciiText.cpp:870: void SoAsciiTextP::setUpGlyphs(SoState*, SoAsciiText*): Assertion `length' failed.
Aborted`

`./FreeCAD-0.17.13541.9948ee4.glibc2.17-x86_64.AppImage 
/bin/bash: /tmp/.mount_FreeCApXoOpD/lib/x86_64-linux-gnu/libtinfo.so.5: no version information available (required by /bin/bash)
desktop-file-validate is missing. Skipping /tmp/.mount_FreeCApXoOpD/usr/bin//freecad.wrapper.
FreeCAD 0.17, Libs: 0.17R13541 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2018
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Geometric Dimensioning & Tolerancing workbench activated
Traceback (most recent call last):
  File "/home/linu3/.FreeCAD/Mod/gdt/GDT.py", line 1171, in updateData
    fp.circumferenceBool = True if (True in [l.Closed for l in fp.faces[0][0].Shape.getElement(fp.faces[0][1]).Edges] and len(fp.faces[0][0].Shape.getElement(fp.faces[0][1]).Vertexes) == 2) else False
<type 'exceptions.TypeError'>: must be string, not tuple
freecad: SoAsciiText.cpp:870: void SoAsciiTextP::setUpGlyphs(SoState*, SoAsciiText*): Assertion `length' failed.
Aborted`

luzpaz, 2019-01-19:
GD&T has not been developed for some time. Perhaps you can open some issues on their repo? It would be really nice to have this WB available for 0.18 as well. 
https://github.com/juanvanyo/FreeCAD-GDT/issues

aversario, 2019-03-12:
With version 0.17 on windows I also get some errors

5axes, 2023-03-24:
Modification to works  : 

`            if (True in [l.Closed for l in fp.faces[0][0].Shape.getElement(fp.faces[0][1][0]).Edges] and len(fp.faces[0][0].Shape.getElement(fp.faces[0][1][0]).Vertexes) == 2) :
                fp.circumferenceBool = True
            else:
                fp.circumferenceBool = False `

https://github.com/5axes/FreeCAD-GDT

luzpaz, 2023-03-24:
@5axes is that a fork of https://github.com/juanvanyo/FreeCAD-GDT/issues ? Should we point the addon manager to your fork then ?

5axes, 2023-03-25:
Yes it's a fork ... but I don't think I have the Python and FreeCad knowledge to support this WB. As FreeCad-GD&T is more a POC than a real useable  WB it's a good start to learn but without the support of other people GD&T will stay in his grave for a long time.

luzpaz, 2023-04-16:
@5axes looks like you've been busy? How do you feel about your fork being the new GD&T addon ?

5axes, 2023-04-16:
No I think it's always better to leave the link to the original source.  Even if I think to continue to work on this project if someone is interested in the project he can always find my fork.  Moreover after having read the last documents concerning the RoadMap of the version 1.0 (https://freecad.github.io/DevelopersHandbook/roadmap/) Especially on the section Code and development I think I am not able to register in the objectives defined by this RoadMap.

5axes, 2023-04-16:
https://github.com/FreeCAD/DevelopersHandbook/issues/34

luzpaz, 2023-04-16:
> Especially on the section Code and development I think I am not able to register in the objectives defined by this RoadMap.

@5axes can you elaborate ?

5axes, 2023-04-16:
https://github.com/FreeCAD/DevelopersHandbook/blob/master/roadmap/index.md#objective--seamless-extensibility

Establish standards for addons to meet to achieve a ‘recommended’ or ‘featured’ status. Featured Addons should

- Have an engaged maintainer   -> _Don't be sure to be an engaged maintainer_
- Have minimal high priority open issues tickets   -> _Have a minimal high priority open issues tickets doesn't mean we are able to solve them_
- Support translation -> _To support translation It lloks like we need to use the QT Development Tools , As non proffesional devellopers don't want to invest in suh tool _
- Support unit schema -> _Don't really undersatnd what they mean by " Support unit schema"_

And 

https://github.com/FreeCAD/DevelopersHandbook/issues/34#issuecomment-1507232292

That would suggest that giving the user the option to use a workbench that doesn't do anything is a bad idea.  ->  _Right now GD&T is a Workbench that doesn't do anything  and it will stay in that level of developpement for a long time_

luzpaz, 2023-04-16:
Thanks. Yes, while i understand why these guidelines are being put in place we also have to acknowledge that FLOSS also slowly comes together over time. I'm still understanding how stringent they must be adhered to because that can also 'strangle' a project when there are high expectations of it. 

So perhaps you can go in to what GD&T does right now ?

5axes, 2023-04-16:
> So perhaps you can go in to what GD&T does right now ?

For the moment it's just a Proof Of Concept.  There is the possibility to create some Reference Plans and Datum Feature, some basic Annotation System and Geometric Tolerance  and a limited Edit function. But to be functional these elements must be upgraded to be fully compatible with the ISO standard.  TED (Theoritical Exact Dimension) are missing (Linear, Angular and Diameter/Radius dimension), as well as Surface finishing . Then we should add some import/export functions (idealy with the STEP AP 242 standard) and link with the TechDraw WB.  to get a full functionnal WB it's certainly some years of work.

chennes, 2024-03-15:
@luzpaz I propose that since the version of GD&T that *is* listed in the Addon Manager is obsolete and non-functional, we mark it as "deprecated" in the [Addon Flags JSON file](https://github.com/FreeCAD/FreeCAD-addons/blob/master/addonflags.json).

5axes, 2024-03-15:
Agree with you : unmaintained and not functionnal