FreeCAD/FreeCAD-addons #117: CuraEngine-Plugin should be removed from FreeCAD-addons

normandc, 2019-02-07:
The [CuraEngine plugin](https://github.com/cblt2l/FreeCAD-CuraEngine-Plugin) has been abandoned since 2014. The newer Cura engine supposedly does not work any more with this plug-in. It apparently stopped working in FreeCAD 0.16. Therefore, there is no logical reason to keep it in the Addon manager, it just generates help requests on the forum for which there is no solution.

Latest topic: https://forum.freecadweb.org/viewtopic.php?f=3&t=34023

luzpaz, 2019-02-07:
Should we have a repo for obsolete addons (that isn't downloadable but is browseable for historical and potential forking ?

yorikvanhavre, 2019-02-07:
I added in the addons manager a list of obsolete add-ons. So far it has only assembly2 in it. What it does is basically add a "Warning: This add-on is marked as obsolete" line in the addon description. An easy thing to do is to add this one to the list at https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/AddonManager/AddonManager.py#L61

But indeed maybe this is a case where it should be fully removed... But removing it from this list also means it will not be uninstallable anymore from the addons manager.

We could create  another repo for these, but this addon repo is actually nothing else than a simple list... The code stays in its original repo. So will still exist anyway.

luzpaz, 2019-05-13:
+1 for creating a repo for obsolete external workbenches

chennes, 2022-01-01:
If we move them, then users of old versions of FreeCAD will no longer have access to them. I think the better solution is to leave them here, but just mark them obsolete. Versions of FreeCAD that don't know about that flag will happily display them, and newer versions will see the flag and indicate it. We may need to extend the "obsolete" flag to include a FreeCAD version or date, though.

chennes, 2022-01-22:
Closing this now that support for per-version deprecation has been merged.

closed 2022-01-22 by chennes