FreeCAD/FreeCAD-addons #120: Addon manager "hangs" while installing big addons

kisolre, 2019-03-11:
With slow connections Addon manager appeart to be stuck while installing big addons (like parts_library - 322MB download). There is no indication of the actual progress and no way to cancel the action so if the user does not have other means to check what is going on (I have a widget that displays download speed graph and also Process explorer) it appears as if the addon manager has hanged.
Forum thread: [https://forum.freecadweb.org/viewtopic.php?f=3&t=32330](url)

OS: Windows 8.1
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16078 (Git)
Build type: Release
Branch: master
Hash: 67df9f3a828fbb0fec29d039792d4baf3d9ee8e0
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: Bulgarian/Bulgaria (bg_BG)

luzpaz, 2019-10-23:
When a user runs `git pull` there is a percentage that is shown while symbols and data is downloaded. This should be shown on the Qt UI progress bar

gz7czq, 2020-08-07:
I am using FreeCAD AppImage v0.18.4 on our OpenSUSE server.  When I select AddOns to download list of addons, it just goes into infinite loop forever.  I need help to either fix the issue or an alternate way to use the DualSPHysics addon.  With the linux AppImage, there is no directory structure for FreeCAD.

chennes, 2022-01-12:
I have modified the UI to display whatever progress information it can: during a git clone operation this is just a bytes-transferred/transfer rate display (not a percent complete), but during a zip download with a size header provided I can provide a percent complete. Implemented as of 1bfbcb5c95d09c166e2f53039ae3ed10b998b6e9.

closed 2022-01-12 by chennes