FreeCAD/FreeCAD-addons #129: Update FreeCAD-Addon-Details.md

luzpaz, 2019-05-17:
This [file](https://github.com/FreeCAD/FreeCAD-addons/blob/master/FreeCAD-Addon-Details.md) is woefully out of date with broken images and links.

limikael, 2019-05-19:
Would maybe the best way to keep it updated be to create a script that updates it automatically? The script could create an entry for each add-on with the following information:

* Name for the add-on (linked to the repo).
* One-liner description (pulled from the repo description).
* A "Details..." link that opens a short description, in the same way as it currently does. However, the details could be automatically pulled from the first X chars (so that it is roughly 5 lines) of the README.md file.
* A "more..." link after the details that point to the repo.

What do you think?

luzpaz, 2019-05-19:
@limikael that sounds frickin' awesome!

luzpaz, 2019-05-19:
and...we could structure the template in https://github.com/FreeCAD/FreeCAD-addons/issues/131 to get the fields the way we want them from submitters

limikael, 2019-05-20:
Interesting! I might give developing such a script a go then... I said "might", so no promise and no commitment, so don't let this stop anyone else who wants to hack... :)

luzpaz, 2019-05-20:
@limikael how would you want to do this? Have a github hook that listens for PRs and then parse the text in the PR ?

limikael, 2019-05-22:
I was actually just thinking a script that can reside in the root of the repository, and it would read the .gitmodules file. It would then loop through the addons, and get the information using the github api and generate FreeCAD-Addon-Details.md. So it wouldn't actually track any PRs. Is there information that will be in the PR that will not be in the repository?

yorikvanhavre, 2019-07-26:
Is maintaining that page still worth it? It's basically duplicating information that is otherwise found - an better explained - on each addon's README. Which is anyway available through the addon manager...

limikael, 2019-08-13:
Good question @yorikvanhavre! I would vote no, it is not worth it.

closed 2019-08-16 by luzpaz