FreeCAD/FreeCAD-addons #130: Create a separate repo or section for outdated/obsolete/unmaintained workbenches/addons

luzpaz, 2019-05-17:
Somehow we should still allow for the Addon-Manager to access said workbenches but the users need to be informed that they have met the above criteria. 

List of Addons to move:  
- [ ] GD&T
- [ ] TBD 
- [ ] TBD

yorikvanhavre, 2019-05-23:
At the moment we maintain a list inside the addon manager itself:
https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/AddonManager/AddonManager.py#L61

Any addon in this list will get an additional message in its description warning the user that the addon is considered obsolete.

We could do something more like separate repos, etc... but it's more work. This is just one line of code :smile:

chennes, 2022-01-01:
The new version of the Addon Manager sort of inverts this, and adds the option to hide these obsolete addons. The option is on by default, so only advanced users who choose to show them will see Py2 and Obsolete addons.

chennes, 2022-01-12:
I have now expanded this idea to include a "deprecated" section in https://github.com/FreeCAD/FreeCAD-addons/blob/master/addonflags.json that allows per-version deprecation (so that for example an addon will continue to work for people with FreeCAD 0.19, but will not show by default for those using 0.20).

closed 2022-01-12 by chennes