FreeCAD/FreeCAD-addons #131: Add a PULL_REQUEST_TEMPLATE.md

luzpaz, 2019-05-17:
Add a similar template like https://github.com/FreeCAD/FreeCAD/blob/master/.github/PULL_REQUEST_TEMPLATE.md

We should ask people to complete several tasks:  
- proper README. Markdown friendly  
- README has a link to a discussion forum for feedback/discussion  
- README has screenshots  
- Py3/Qt5 compatible
- Run `codespell` to remove typos
- more ?

gam271, 2019-08-16:
@luzpaz I'd love to help if you still need it!

luzpaz, 2019-08-16:
@gam271 great. I'll let you know if I think of something. If you see something, feel free to propose your view/change. :+1:

gam271, 2019-08-16:
Hello again @luzpaz I realized that on your Changelog.txt all the capitalization is not consistent and I'd be happy to help with that.  and on the Readme.md in the overview section, I would suggest adding a period after the bold text.

luzpaz, 2019-08-16:
@gam271 what `Changelog.txt` file?  
Go ahead and create your PR and I'll review it. Thanks!

luzpaz, 2019-09-16:
homebrew-cask has a ingenious way of making their template easy to create:  
```
Additionally, if **adding a new cask**:

- [x] Named the cask according to the [token reference].
- [ ] `brew cask install {{cask_file}}` worked successfully.
- [ ] `brew cask uninstall {{cask_file}}` worked successfully.
- [x] Checked there are no [open pull requests] for the same cask.
- [x] Checked the cask was not [already refused].
- [x] Checked the cask is submitted to [the correct repo].
- [x] Attempted to find an [appcast].

[token reference]: https://github.com/Homebrew/homebrew-cask/blob/master/doc/cask_language_reference/token_reference.md
[open pull requests]: https://github.com/Homebrew/homebrew-cask/pulls
[already refused]: https://github.com/Homebrew/homebrew-cask/search?q=is%3Aclosed&type=Issues
[the correct repo]: https://github.com/Homebrew/homebrew-cask/blob/master/doc/development/adding_a_cask.md#finding-a-home-for-your-cask
[version-checksum]: https://github.com/Homebrew/homebrew-cask/blob/master/doc/cask_language_reference/stanzas/sha256.md#updating-the-sha256
[appcast]: https://github.com/Homebrew/homebrew-cask/blob/master/doc/cask_language_reference/stanzas/appcast.md
```

Apparently this works:
```markdown
- [ ] Some generic sentence with a [link].

[link]: https://freecadweb.org
```
:point_up_2:
**Note:** a newline is needed between these 2 statements

**Resulting in**:  
- [ ] Some generic sentence with a [link].

[link]: https://freecadweb.org

commit 7cb4a103 (2019-10-10): Update issue templates

Experimenting with an issue template.  
Closes #131
commit 77428c40 (2019-10-10): Update issue templates

Experimenting with an issue template.  
Closes #131