FreeCAD/FreeCAD-addons #138: MOOC WB incorrectly added

luzpaz, 2019-07-12:
Hey @yorikvanhavre I tried to add the MOOC workbench and it didn't actually make a submodule that I'm familiar with. It's because I tried a different terminal command this time:  
`git submodule add --reference https://framagit.org/freecad-france/mooc-workbench --force ./MOOC/ `
because `git submodule add MOOC https://framagit.org/freecad-france/mooc-workbench.git` wouldn't work

yorikvanhavre, 2019-07-12:
Fixed in 8f9815e
The correct syntax is `git submodule add https://framagit.org/freecad-france/mooc-workbench.git MOOC`

yorikvanhavre, 2019-07-12:
Hmm there is a problem, though. As it is not hosted on github, github refuses to show a link to it on the main page... So the AddonManager won't be able to fetch it. I'll add an exception to it for now in the addon manager code, but we'll  need a better solution

wood-galaxy, 2019-07-12:
I get an error when start the addons manager : 
`Traceback (most recent call last):
  File "/usr/share/freecad-daily/Mod/AddonManager/addonmanager_workers.py", line 89, in run
    url = "https://github.com/" + re.findall("href=\"\/(.*?)\/tree",l)[0]
IndexError: list index out of range
`
With 0.18.2 and 0.19 both with Py3Qt5

BTW I can make a mirror on github for the MOOC workbench

yorikvanhavre, 2019-07-12:
No, let's solve it... We need to support non-github repos, this is a very good test. Also, since framagit uses gitlab, this will serve for other gitlab instances too...
I'll push a fix for the addon manager in a minute

sgrogan, 2019-07-12:
> No, let's solve it... We need to support non-github repos, this is a very good test.
+1, you are the best

yorikvanhavre, 2019-07-12:
Done https://github.com/FreeCAD/FreeCAD/commit/e5c823bf34fdcf5a0fe34696c9d02aceda9f3cea
The README is still not readable... Not sure what to do there.

wood-galaxy, 2019-07-12:
I have not tried it.
But addon manager is still unusable on 0.18.2 with the error above

yorikvanhavre, 2019-07-12:
I didn't backport to 0.18

wood-galaxy, 2019-07-12:
Well, I completely agree that we have to support other git instances than Github. But I'm afraid that's impossible before version 0.19...
I therefore propose to add this mirror in the meantime: https://github.com/wood-galaxy/mooc-workbench

luzpaz, 2019-07-12:
@yorikvanhavre feel free to change the submodule link to https://github.com/wood-galaxy/mooc-workbench

yorikvanhavre, 2019-07-12:
In https://github.com/FreeCAD/FreeCAD/commit/3129ae4296e40ed20e7b3d460b86e6969acbe1c3 I fixed 0.18 so at least it doesn't crash. But it won't add the non-github workbenches.

Ok I'll edit the URL

wood-galaxy, 2019-07-12:
Thanks @yorikvanhavre !
I will wait 0.19 to going back to the framagit instance ;)

closed 2019-07-12 by wood-galaxy

yorikvanhavre, 2019-07-12:
That's a good thing with git.. One more mirror is not much work.. :)
In my freecad repo I configured it to push to several remote repos when I "git push" (need to set several "push-url")