FreeCAD/FreeCAD-addons #148: "addons_installer.FCMacro" file link "404 not found" error

MantaRayDeeJay, 2019-08-19:
Hello,
I would like to download the "addons_installer.FCMacro" file but it seems that there is a "404 not found" error.

[image]

luzpaz, 2019-08-19:
What for? It's deprecated.  
See https://github.com/FreeCAD/FreeCAD-addons/issues/141

luzpaz, 2019-08-19:
Unless you're using an inferior version of FreeCAD, we recommend using the built-in Addon Manager to download workbenches and macros. Nonetheless, the link tot he deprecated macro is now fixed.  
Closing ticket

closed 2019-08-19 by luzpaz

MantaRayDeeJay, 2019-08-19:
I'm always using the v0.16 of FreeCAD because I can use it on a Windows XP Light Weight VM guest.