FreeCAD/FreeCAD-addons #151: Support develop-branches / releases of 3rd-party modules

looooo, 2019-09-30:
Would be nice to somehow support different version of 3rd-party modules. I guess this would be useful for:
1. 3rd-party modules dependent on specific versions of freecad. In this case the addon-manager should install a version of the 3rd-party module which works with the version of freecad.
2. Install and test specific features which are developed in a seperate branch (keeps master-branch stable, avoids backward incompatibilities.)

Idea came up in this issue:
https://github.com/looooo/freecad.gears/issues/1

sgrogan, 2019-10-01:
Interesting.
For 2, it seems possible to do manually.
I tested with the Gears WB that was install with the add-on manager using git-python.
I bashed into %appdata%/FreeCAD/Mod/FCGear
I was able to git checkout develop
I was also able to git checkout -b test
I changed the version in pygears/_init_.py, and after a FreeCAD re-start the change is picked up in the FreeCAD python console "print(pygears.__version__)"

I guess the challenge would be all the visual stuff so the user knows they are not at HEAD of master or on a branch (maybe also if the branch is local, not in upstream)

chennes, 2022-01-01:
Part of this is on my radar (at least, adding support for multiple git branches/refs, and switching between them via the GUI). I probably can't easily support having multiple versions of the same WB installed at the same time, though.

chennes, 2022-02-21:
I've merged that branch-switching PR -- I think that between the new package.xml metadata's ability to set FreeCAD version information, and the new ability to switch between branches, we've covered the main thrust of this feature request.

closed 2022-02-21 by chennes