FreeCAD/FreeCAD-addons #155: [FEATURE] Make OpenSCAD WB able to function as an External Workbench

luzpaz, 2019-10-17:
Related forum thread: https://forum.freecadweb.org/viewtopic.php?f=42&t=40085&start=10#p340987

luzpaz, 2019-10-22:
CC @KeithSloan @looooo  
Based on the forum thread and acknowledging that OpenSCAD is important within FC, there is still an interest to have it operate as an external workbench for several reasons. One very important one is that there could be more flexibility adding features and maintaining (since an addon doesn't need to be committed in to master)

luzpaz, 2020-01-19:
Disucssion about removing PLY from FC source code (and make it an external dependency): https://forum.freecadweb.org/viewtopic.php?f=10&t=42476

vocx-fc, 2020-02-10:
[OpenSCAD as installable workbench](https://forum.freecadweb.org/viewtopic.php?f=10&t=43206)

PhoneDroid, 2026-04-02:
Stale

closed 2026-04-02 by PhoneDroid