FreeCAD/FreeCAD-addons #159: [FEATURE] Skipping restart with this git trick ?

luzpaz, 2019-10-23:
Per @sgrogan in https://github.com/looooo/freecad.gears/issues/1#issuecomment-537256864

> I think something like this, from FreeCAD python console, should work?
> 
> ```python
> import git
> repo = git.Repo(FreeCAD.getUserAppDataDir()+'/Mod/FCGear')
> repo.git.pull('origin','develop')
> repo.git.checkout('develop')
> import importlib
> importlib.reload(pygears)
> ```
> 
> EDIT: no restart required PY3.4+

sgrogan, 2019-10-24:
only the
 `import importlib
importlib.reload(pygears)`
is necessary. The other stuff is to checkout a development branch of the repo instead of master.

luzpaz, 2020-02-06:
Related: https://forum.freecadweb.org/viewtopic.php?f=10&t=29805

luzpaz, 2022-02-16:
@wmayer does 0bbc253ddee6801770c78e011c5201b7c2c36080 have any relevance to this feature request? (ignoring the 'git trick' referenced in this ticket, I mean the ability to not requiring a FreeCAD restart)

wwmayer, 2022-02-17:
No.