FreeCAD/FreeCAD-addons #161: [FEATURE] If addons/macros not available (AKA 404) show a more friendly/utilitarian message

luzpaz, 2019-12-09:
**Is your feature request related to a problem? Please describe.**  
https://github.com/FreeCAD/FreeCAD-macros/issues/68  
freecadweb.org was down, user didn't understand why.

**Describe the solution you'd like**  
Perhaps create a dialog or a clearer message as to what might be going on and how to fix it.  
```text
It seems the website and/or online repository where addons/macros are retrieved from is not available. Possible issues:  
Is this machine still connected to the internet?  
Is this machine behind a proxy?  
Is the website (not sure if we should add freecadweb.org for branding reasons) available?  
If none of the above are relevant, please discuss this issue on the FreeCAD community forum (https://forum.freecadweb.org/)

**Describe alternatives you've considered**
N/A

**Additional context**
The original issue occurred because freecadweb.org was unavailable.

luzpaz, 2020-09-13:
For macros: https://github.com/FreeCAD/FreeCAD/commit/dee22fab8608979f1e4491d654dd495998f8dbd9

chennes, 2022-02-27:
This particular issue really only affected macros (that is, our domain being inaccessible). Our other main source is GitHub, and the recent updates to the Addon Manager are much clearer about a lack of connection there. I think this can be closed now -- there's always room to improve error messaging, but it's in a pretty solid place at the moment.

closed 2022-02-27 by luzpaz