FreeCAD/FreeCAD-addons #163: [FEATURE] Ask Addon Authors to use LGTM CI analysis

luzpaz, 2020-01-19:
**Is your feature request related to a problem? Please describe.**  
Standardizing code quality in all Addons 

**Describe the solution you'd like**  
Using [LGTM](https://www.freecadweb.org/wiki/LGTM) 

**Additional context**  
Asking Authors to add a `.lgtm.yml` file and sign up for their free service to run a check on their specific Addon repo so it generates a report of issues to be fixed.

luzpaz, 2020-01-19:
Related ticket: https://github.com/FreeCAD/FreeCAD-macros/issues/73

Thovthe, 2022-04-06:
Deafeaturing should get a checkmark.

luzpaz, 2023-01-05:
LGTM has now been sunset as a service: https://github.blog/2022-08-15-the-next-step-for-lgtm-com-github-code-scanning/

closed 2023-01-05 by luzpaz