FreeCAD/FreeCAD-addons #164: Purpose of adding a new workbench to a template in the wiki

vocx-fc, 2020-01-26:
The current instructions to add a new external workbench to this FreeCAD/FreeCAD-addons repository say this.

>3. Create an entry on https://www.freecadweb.org/wiki/Template:DevWorkbenches

This page is a "Template", which has a specific meaning in Mediawiki software. See https://www.mediawiki.org/wiki/Help:Templates.

Essentially, a template is meant to be included in many pages in the wiki. If the page `Template:My_template` is defined, it can be included in other pages like this `{{My_template}}`.

So, my question is, why should the authors of external workbenches add their new workbench specifically to a template page? Why should they add it to `Template:DevWorkbenches`? This only makes sense if this page is used somewhere else, like `{{DevWorkbenches}}`.

I don't think it makes sense to create a template with all the external workbenches. All, or most, are listed and categorized in the page https://www.freecadweb.org/wiki/external_workbenches.

And to clarify, I do think it makes sense to list all workbenches in a nice table, with author and website information, I just don't think this page should be a template. I mean, it should be a regular wiki page.

So, the instruction should say something like the following.

>3. Create an entry on https://www.freecadweb.org/wiki/External_workbench_list

This page would have the same information as the current `Template:DevWorkbenches` page but it wouldn't be a "template", it would be a regular page.

I hope I am explaining myself well. Basically, it is fine to have the information in a table, but there is no point in this information being inside a "template".

Also, about the current `Template:DevWorkbenches` page, I don't get why the information is repeated, once in a table, and once in a simple bulleted list. I feel this page is just some archaic page that was created long ago, and was just left like that.

vocx-fc, 2020-01-26:
I created the wiki page https://www.freecadweb.org/wiki/External_workbench_list with my chosen organization. So, in my opinion, the instructions should say to add a new workbench to this page, and not to the `Template:DevWorkbenches` page.

The [External_workbench_list](https://www.freecadweb.org/wiki/External_workbench_list) page currently lists all the workbenches available from the Addon Manager, in an alphabetical ordering.

I think it makes sense to also have it in a thematic order, just like the sections in https://www.freecadweb.org/wiki/External_workbenches. So, I'm thinking that perhaps we only need one list that combines `External_workbenches` with `External_workbench_list`.

vocx-fc, 2020-02-09:
I finished re-organizing the external workbenches pages in the wiki.

I suggest changing the instructions as follows:

1. Announce your workbench in the FreeCAD forums in the [Announcement section](https://forum.freecadweb.org/viewforum.php?f=9). Also create a discussion page where you will provide support for your workbench, in the [Open Discussion section](https://forum.freecadweb.org/viewforum.php?f=8).
2. Create a dedicated page for your workbench on the FreeCAD wiki, for example, [MyOwn_Workbench](https://freecadweb.org/wiki/MyOwn_Workbench). In this page include `[[Category:Addons]]` at the bottom. Add your workbench, and its wiki page, to the list of [External workbenches](https://freecadweb.org/wiki/External_workbenches), placing it in the proper section.
3. After your workbench is accepted into the addons repository, it will be included in the [External workbench list](https://freecadweb.org/wiki/External_workbench_list).

The rest of the points stay the same.

My idea is that the page "External_workbenches" contains every single workbench that is there, while "External_workbench_list" contains only those that appear in fact in the Addon Manager. We must definitely stop using the "Template:DevWorkbenches" page as it serves no purpose.

What do you think @luzpaz?

Can you commit these changes or should I open a pull request?

chennes, 2024-03-15:
I'm closing this as completed and/or obsolete now.

closed 2024-03-15 by chennes