FreeCAD/FreeCAD-addons #169: [New Addon/Workbench] Geomatics is deprecated, replaced with freecad.trails

elliotclee, 2020-06-08:
Hi,

The note on the repository of the Geomatics workbench says it has been deprecated/abandoned and merged into https://github.com/joelgraff/freecad.trails . Please consider updating the addons list to match.

yorikvanhavre, 2020-06-19:
Okay trails (and pivy_trackers) are added to the addons list! ( ping @joelgraff @HakanSeven12 )

closed 2020-06-19 by yorikvanhavre

elliotclee, 2020-06-25:
I don't think pivy_trackers is a workbench. Instead, the trails workbench should be checked out with --recursive so that pivy_trackers is included with the trails checkout.

elliotclee, 2020-06-25:
Also, the addons list is pretty much empty right now in FreeCAD 0.19.21718 on Windows.

joelgraff, 2020-06-25:
> I don't think pivy_trackers is a workbench. Instead, the trails workbench should be checked out with --recursive so that pivy_trackers is included with the trails checkout.

Correct.  I've incorporated pivy_trackers as a submodule in Trails to avoid unnecessary work for the user installing.  Though submodules have provided their own share of headaches so far...